boltgolt/howdy #625: initial login freezes up laptop with howdy enabled

alwzfolwd, 2021-12-19:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._
im running,

 Operating System: Kali GNU/Linux 2021.4
KDE Plasma Version: 5.23.4
KDE Frameworks Version: 5.88.0
Qt Version: 5.15.2
Kernel Version: 5.14.0-kali4-amd64 (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i3-1005G1 CPU @ 1.20GHz
Memory: 7.3 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics

             I was able to install and configure howdy just fine and it works just fine for any authentication other than the login screen. Upon entering the login screen howdy does not scan for a face so i enter my password instead. After entering my password howdy then initiates a face scan (i know because my webcam light turns on when howdy is started). The screen then freezes up and im not able to do anything. The only way i was able to log into my pc was to open a terminal from the login screen and disable howdy. The funny thing is that in order to get into the terminal i had to authenticate and howdy initialized and logged me in no problem. So, to me it seems to be a login screen issue, im currently using sddm, im not sure if that makes a difference or not. I'm probably wrong, just a hypothesis. 
    
            I tried sudo chmod o+x /lib/security/howdy/dlib-data/ from issue #431 with no errors and still the problem persists.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):Operating System: Kali GNU/Linux 2021.4

Howdy version (`sudo howdy version`):Howdy 2.6.1

dovepa, 2021-12-26:
same bug on dell xps 9510

EmixamPP, 2021-12-30:
`sudo pacman -S fmt` ?

WertsMarc, 2022-02-04:
same here, Dell XPS 13 9310

boltgolt, 2022-02-06:
Anything logged in `/var/log/auth.log` for any of you?

boltgolt, 2022-02-07:
Log looks good, howdy does what it should. Any modifications to the lockscreen?

adelavega, 2022-09-01:
+1 on this issue. The weight thing is this didn't used to happen, and now there is a long delay or it requires a password to be typed before it tries to launch howdy

techonerd, 2023-11-20:
+1 to this issue. Facing this on Kali Linux. It works fine for sudo auth. But for login, it asks for a password and then initiates the face recognition. I am able to see the snaps of login in `/usr/lib/security/howdy/snapshots`

closed 2026-11-16 by boltgolt