boltgolt/howdy #629: When switching user, logging in does not work (sddm)

lnee94, 2022-01-08:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

lnee94, 2022-01-08:
note tested in a vm with kubuntu 21.10

lnee94, 2022-01-20:
So, a problem with kubuntu?

boltgolt, 2022-01-20:
It almost always is in these cases. What could help is changing the permissions on the Howdy folder to be world readable

Aetf, 2022-01-21:
Actually in this case I believe this isn't the issue with PAM setup. It's about a wrong assumption on the Howdy side: the pam stack is not always run as root.

For example, the plasma screen locker uses `kcheckpass`, which then invokes pam to do the actual checking. When `kcheckpass` is linked with PAM, it is just a normal program [without SUID bit set](https://invent.kde.org/plasma/kscreenlocker/-/blob/master/kcheckpass/CMakeLists.txt#L17-L18), so it will run as the normal user it's invoked from.

See [this](https://stackoverflow.com/a/24152223) for more details about the user the PAM stack is running as.

lnee94, 2022-02-02:
what do i do.

sneakyjoeru, 2023-02-21:
See if it also happens if you just lock screen and can't use it. I applied 755 permissions to howdy and adjusted 2 config files.
https://github.com/boltgolt/howdy/issues/219#issuecomment-674606116 - this may help, if it's the same issue.

`$sudo chmod -R 755 /lib/security/howdy`
insert `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/kde` and `/etc/pam.d/system-auth`

closed 2026-11-16 by boltgolt