boltgolt/howdy #630: Integration of howdy with gnome polkit doesn't work

Itotutona, 2022-01-13:
Hello,

I tryed to add howdy to polkit but it doesn't work, could you help me please ?

> /etc/pam.d/polkit-1

```#%PAM-1.0
auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
auth       include      system-auth
account    include      system-auth
password   include      system-auth
session    include      system-auth
```

Linux distribution: Fedora 35

Howdy version: 2.6.1

ad-on-is, 2022-01-15:
Just installed  howdy with GNOME 41.3 and it works.

Although, my line looks different than yours.

```
auth       sufficient   pam_python.so /lib/security/howdy/pam.py
```

`lib` instead of `lib64`

Itotutona, 2022-01-16:
Are you using Fedora ? I use /lib64/security/howdy/pam.py because I follow the documentation in the copr

ad-on-is, 2022-01-16:
I'm on EndeavourOS, running vanilla GNOME

Itotutona, 2022-01-16:
My problem is I never see polkit. My webcam is active (flash is active) but I don't see any popup (by default polkit open a popup). And until I kill the application, my camera is active but I can't connect.

ad-on-is, 2022-01-16:
Aaah, did you check the howdy config.ini. There's some setting about showing a notification.

ad-on-is, 2022-01-16:
And btw, your title is misleading, since everyone would assume it's not working at all. Just try to be more precise.

ad-on-is, 2022-01-18:
Sorry, my bad, didn't know how Fedora handles it.

nickavem, 2022-03-20:
I've been troubleshooting this, and it doesn't seem to be setting off any SELinux alarms. Getting same behavior, though.

Itotutona, 2022-03-20:
I works on [Arch Linux](https://archlinux.org) with the howdy package.

EmixamPP, 2022-07-14:
I can confirm that this is not due to SELinux.

Then I investigated a bit, and it appears that when authenticating with polkit, it "spam" the authentication request and kill the previous process and so it never gets the response from Howdy which was comparing the face read by the camera to the model. 

If I modify the pam.py script that handles pam, to reply that the authentication is approved before calling Howdy, then there is no problem, pam doesn't call the authentication script again.

So the problem lies in the polkit configuration.

EmixamPP, 2022-07-14:
I continued my tests and the problem is not that polkit is "impatient"; if I execute in the Howdy's pam script a simple "sleep 4" using a shell subprocess, polkit waits for the end of the 4 seconds. 

I have two hypotheses to check, the Howdy's comparison script
1. is not allowed to run and so the process is killed and recalled
2. crashes for some reason and it is recalled

The second point seems easy to check, I'll do that another time. Also, let me know if I'm sending a bit too much message, but I'd rather detail what eventually someone finds the solution.

jamesdangercarpenter, 2022-07-15:
Just a +1, I'm also experiencing this, and ```setenforce 0``` didn't help... Thanks for the updates EmixamPP there are useful for me. I'll try and investigate from my side also and update if I find anything useful.

TenSeventy7, 2022-07-16:
Could confirm that #687 works (pretty well) on my Fedora install.