boltgolt/howdy #631: Not working at lock screen

aloksethi, 2022-01-26:
howdy is working ok for sudo but whenever I lock the screen and try to unlock, it doesn´t. The red light next to the IR webcam doesn´t blink either.

I have set execute  permissions for all on both 
`/lib/security/howdy`
`/usr/lib/security/howdy/`

Below are the logs from auth.log for  an unsuccessful case (lock screen gnome-screensaver)
```
Jan 26 23:53:50 asethi-TUXEDO /lib/security/howdy/pam.py[5613]: Traceback (most recent call last):
Jan 26 23:53:50 asethi-TUXEDO /lib/security/howdy/pam.py[5613]:   File "/lib/security/howdy/pam.py", line 93, in pam_sm_authenticate
Jan 26 23:53:50 asethi-TUXEDO /lib/security/howdy/pam.py[5613]:     return doAuth(pamh)
Jan 26 23:53:50 asethi-TUXEDO /lib/security/howdy/pam.py[5613]:   File "/lib/security/howdy/pam.py", line 39, in doAuth
Jan 26 23:53:50 asethi-TUXEDO /lib/security/howdy/pam.py[5613]:     pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Attempting face detection"))
Jan 26 23:53:50 asethi-TUXEDO /lib/security/howdy/pam.py[5613]: PamException: Conversation error
Jan 26 23:54:15 asethi-TUXEDO gnome-screensaver-dialog[5613]: gkr-pam: unlocked login keyring
```

 and a successful case (running sudo on command line).
```
TJan 26 23:54:55 asethi-TUXEDO [HOWDY]: Attempting facial authentication for user asethi
Jan 26 23:54:56 asethi-TUXEDO [HOWDY]: Login approved
Jan 26 23:54:56 asethi-TUXEDO sudo:   asethi : TTY=pts/2 ; PWD=/usr/lib/security/howdy ; USER=root ; COMMAND=/bin/bash
Jan 26 23:54:56 asethi-TUXEDO sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
```

Linux distribution (if applicable): Ubuntu 20.04.3 LTS

Howdy version (`sudo howdy version`): 2.6.1

aloksethi, 2022-01-27:
`no_confirmation` was already true. However, I had to turn `detection_notice` to false too and after that it worked.

unfortunately, it seems i myself had turned `detection_notice` to true in the first place.

# Print that face detection is being attempted
detection_notice = false

# Do not print anything when a face verification succeeds
no_confirmation = true

Euroaborigenas, 2023-01-20:
> 

dude, I owe you a beer.
Whole day today I was doing all the possible suggested manipulations around internet to get it work on my Budgie locked screen activation. Nothing helped. 
THANK YOU!!! 🍺