boltgolt/howdy #633: Cannot get PAM to use Howdy

LukeFernandes, 2022-02-06:
Cannot get PAM to use Howdy -

Could be a Python 3 issue as only appeared after the Manjaro December upgrade to Python 3.

At first, the error was:

```
PAM adding faulty module: /usr/lib/security/pam_python.so
pkexec[3912]: PAM unable to dlopen(/usr/lib/security/pam_python.so): /usr/lib/security/pam_python.so: cannot open shared object file: No such file or directory

```
Now, after rebuilding python-dlib and pam-python, that error seems to have gone, but I get:
```
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]: Traceback (most recent call last):
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/lib/security/howdy/pam.py", line 93, in pam_sm_authenticate
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     return doAuth(pamh)
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/lib/security/howdy/pam.py", line 30, in doAuth
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     if config.getboolean("core", "ignore_closed_lid"):
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/usr/lib/python2.7/ConfigParser.py", line 368, in getboolean
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     v = self.get(section, option)
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/usr/lib/python2.7/ConfigParser.py", line 618, in get
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     raise NoOptionError(option, section)
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]: NoOptionError: No option 'ignore_closed_lid' in section: 'core'
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]: Traceback (most recent call last):
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/lib/security/howdy/pam.py", line 93, in pam_sm_authenticate
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     return doAuth(pamh)
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/lib/security/howdy/pam.py", line 30, in doAuth
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     if config.getboolean("core", "ignore_closed_lid"):
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/usr/lib/python2.7/ConfigParser.py", line 368, in getboolean
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     v = self.get(section, option)
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:   File "/usr/lib/python2.7/ConfigParser.py", line 618, in get
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]:     raise NoOptionError(option, section)
Feb 06 11:38:07 Putty4-7manjaro /lib/security/howdy/pam.py[2043]: NoOptionError: No option 'ignore_closed_lid' in section: 'core'

```
Howdy test works absolutely fine.

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Manjaro, Linux kernel 5.10, Cinnamon

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2022-02-06:
Interestingly you have a config version missmatch, the option `ignore_closed_lid` is getting replaced with `abort_if_lid_closed` in the unreleased 3.0.0 version. What does your config file look like? What happens when you use a [clean config](https://github.com/boltgolt/howdy/blob/v2.6.1/src/config.ini) from 2.6.1?

LukeFernandes, 2022-02-06:
Hi boltgolt, thanks for the quick reply. I took a look at my config file and sure enough, there is a 'pacnew' file next to it which is more up to date. Pacman must have downloaded the new config file but not made it the actual config - so I tweaked the new one to match my existing (e.g. width and height settings) and made it the active config.ini. It works now! :)

boltgolt, 2022-02-06:
Good to hear!

closed 2022-02-06 by boltgolt