boltgolt/howdy #634: KDE Wallet question

dm30, 2022-02-06:
Hi. During installation KDE Wallet asks for creating a 'wallet' but tells a password was not set. As I'm using Ubuntu, the password manager is GNOME's and seems its not checked. But I can use Howdy normally. Can that be an issue?

I took some snapshots (excuse they are in spanish. I hope its easily understandable)
Thank you.

[image]
[image]
u

boltgolt, 2022-02-06:
Problem with Howdy is that it can't unlock your keychain as you never enter your password, which is used to unencrypt your keychain. One solution would be to not use Howdy for your login

closed 2022-02-06 by boltgolt

nalin29, 2022-02-06:
Alternatively you could set the keychain to empty then it will automatically unlock. But that is living dangerously.

dm30, 2022-02-06:
Thank you. But I don't understand. Is this normal behavior or I shouldn't care? Howdy seems to work normally.

boltgolt, 2022-02-06:
Basically there's a concept in Ubuntu (and other distros of course) called a "keychain" that contains secret passwords and is encrypted on disk. To unlock this chain and use the secrets it needs you login password. When you login with Howdy you don't provide your password at all and thus your keychain can't be unlocked. 

If that's fine with you you can keep it like this. To fix it you can disable Howdy for logging in, or live your life very dangerously and set an empty password for you keychain as @nalin29 said

nalin29, 2022-02-06:
Only if it bothers you it’s normal behavior. But you would have to unlock the keychain for certain things such as Wi-Fi passwords.