boltgolt/howdy #635: OBS freezes with Howdy enabled

uxkjaer, 2022-02-08:
When using OBS you need the camera, if I try to start the virtual camera it will prompt me for a password. If I have howdy enabled, then OBS freezes as it tries to access the same camera.

Any way to tell howdy to be disabled for certain programs or maybe a fallback to password?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Zorin OS 16

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2022-02-09:
Howdy does fall back to a password if facial authentication fails, but I guess it all crashes before that can happen. Check the wiki on here for a short article on how to only use Howdy in certain places

closed 2022-02-09 by boltgolt

uxkjaer, 2022-02-10:
Thanks for the quick reply. I checked the wiki, that only tells me how to disable it as a pam module and then manually set it for sudo commands only. Is it possible in the config to say exclude from these apps?

Sorry if i've missed something.

boltgolt, 2022-02-21:
I'm going to mark this as wontfix as there's not really a solution to 2 applications using the same camera