boltgolt/howdy #637: Cannot dismiss KDE lockscreen

emeraldtip, 2022-02-19:
So howdy works just fine with sudo, but I can't use it to dismiss my lockscreen on KDE

In journalctl this pops up:
`Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall /lib/security/howdy[1923]: <type 'exceptions.SyntaxError'>: ('I/O error while reading', ('/lib/security/howdy', 1, 0, ''))
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall dbus-daemon[575]: [system] Activating via systemd: service name='org.freedesktop.home1' unit='dbus-org.freedesktop.home1.service' requested by ':1.77' (uid=1000>
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall dbus-daemon[575]: [system] Activation via systemd failed for unit 'dbus-org.freedesktop.home1.service': Unit dbus-org.freedesktop.home1.service not found.
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall kcheckpass[1923]: pam_systemd_home(kde:auth): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall audit[1926]: USER_AUTH pid=1926 uid=1000 auid=1000 ses=2 msg='op=PAM:unix_chkpwd acct="emerald" exe="/usr/bin/unix_chkpwd" hostname=? addr=? terminal=? res=fail>
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall unix_chkpwd[1926]: password check failed for user (emerald)
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall kcheckpass[1923]: pam_unix(kde:auth): authentication failure; logname= uid=1000 euid=1000 tty=:0 ruser= rhost=  user=emerald
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall kcheckpass[1923]: Authentication failure for emerald (invoked by uid 1000)
Feb 19 10:45:45 minecraftsexmodhalal2020freeinstall kernel: audit: type=1100 audit(1645260345.156:118): pid=1926 uid=1000 auid=1000 ses=2 msg='op=PAM:unix_chkpwd acct="emerald" exe="/usr/bin/unix_chkpwd" hostname>
Feb 19 10:45:48 minecraftsexmodhalal2020freeinstall kscreenlocker_greet[1893]: file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/SessionManagementScreen.qml:76:9: QML Label: Bindin>
Feb 19 10:45:48 minecraftsexmodhalal2020freeinstall kscreenlocker_greet[1893]: file:///usr/share/plasma/look-and-feel/org.kde.breeze.desktop/contents/components/SessionManagementScreen.qml:76:9: QML Label: Bindin>
Feb 19 10:45:51 minecraftsexmodhalal2020freeinstall [1928]: Attempting facial authentication for user emerald
Feb 19 10:45:51 minecraftsexmodhalal2020freeinstall [1928]: Failure, unknown error1`

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution: Arch

Howdy version: 2.6.1

I really hope it's not just me being an idiot

boltgolt, 2022-02-19:
First of all, great hostname for your machine. Second, this is a very common issue with loads of threads about it and even a [wiki page](https://github.com/boltgolt/howdy/wiki/Common-issues#lockscreen-not-unlocking). Please use the search function

closed 2022-02-19 by boltgolt

emeraldtip, 2022-02-22:
I have tried all those steps, I have tried things from other issues, but none of them work