boltgolt/howdy #638: Unable to install howdy under Ubuntu 21.10

sushan1223, 2022-02-20:
[image]

boltgolt, 2022-02-21:
Seems like you have a very broken package registry, can you run `sudo apt-get update --fix-missing`? Also, please paste console output into github in stead of a screenshot so others can also find this issue too

harivamsi9, 2022-03-22:
**Tested on Ubuntu 21.10, Installation was success.**
Followed the instructions given in README.md file

`sudo add-apt-repository ppa:boltgolt/howdy
`
`sudo apt update
`
`sudo apt install howdy`
Attaching screenshots for your reference - 
[image]

[image]

[image]

[image]

Text File of Output for reference:
[howdy_installation.txt](https://github.com/boltgolt/howdy/files/8327144/howdy_installation.txt)

harivamsi9, 2022-03-22:
@sushan1223 usually 

> <package> but is not installable
these occur when the <package> itself is not present in the ubuntu universal repo, I checked the repo, and these dependencies which ever were shown in your output are all present and compatible only.
So the only explanation is that you may have to update `sudo apt update` and fix broken `sudo apt-get update --fix-missing`
these should fix.

boltgolt, 2022-04-18:
No response from OP, seems like a broken registry

closed 2022-04-18 by boltgolt