boltgolt/howdy #641: Howdy fails to open IR camera

theLMGN, 2022-03-12:
```
Adding face model for the user foxt
Enter a label for this new model [Initial model] (max 24 characters): 
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/videoio_registry.cpp (217) VideoBackendRegistry VIDEOIO: Enabled backends(8, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940); UEYE(930)
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/backend_plugin.cpp (369) getPluginCandidates VideoIO plugin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/backend_plugin.cpp (376) getPluginCandidates     - /lib64: 0
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/backend_plugin.cpp (380) getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (831) open OpenCV | GStreamer: /dev/video1
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (864) open OpenCV | GStreamer: mode - FILE
[ WARN:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (1034) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (597) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
MFX: Can't initialize session
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_images.cpp (282) icvExtractPattern Pattern: /dev/video%01d @ 1
[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_images.cpp (313) open CAP_IMAGES: Stop scanning. Can't read image file: /dev/video1

Please look straight into the camera
Failed to read camera specified in your 'device_path', aborting```

Switching to video0 makes it work, but obviously that's just using RGB camera.

---

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora Workstation 35

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2022-03-12:
Does your IR camera work in other applications? Like zcheese for example

theLMGN, 2022-03-12:
Unsure what zcheese is, but in Cheese both devices use the RGB.

theLMGN, 2022-03-12:
Under Windows 11, the camera is detected as being a regular RGB camera and uses the regular UVC camera driver, and therefore no Windows Hello.

Under Windows 10 1809, a Realtek driver is installed and Windows Hello works fine

Under Linux the `usbfs` driver is used. I remember using this camera with Howdy many years ago under an older Ubuntu version (maybe 18.10, 19.04 or 19.10? not sure)

```
Bus 001 Device 015: ID 0408:7020 Quanta Computer, Inc. RGB-IR Camera
```
```
    |__ Port 8: Dev 15, If 1, Class=Video, Driver=usbfs, 480M
    |__ Port 8: Dev 15, If 0, Class=Video, Driver=usbfs, 480M

```

boltgolt, 2022-04-18:
Sorry, for that typo, did mean Cheese. If both available cameras in Cheese are in color then the Linux kernel does not have drivers for your IR camera unfortunately, maybe possible to downgrade that if it worked before but that seems risky

cannondale0815, 2022-04-24:
Just coming here to share the experience I had with my Logitech Brio camera. I couldn't get IR to work until I updated the firmware of the camera to the latest available version using the "Logi Tune" application on a Windows PC.

Check if there's a firmware update available for your camera.

Before the firmware update, I only had /dev/video0 and /dev/video1 available. After, I now also have video2 and video3. And using video2 worked for IR functionaliy.

Perhaps this comment will help someone else.

closed 2022-04-28 by boltgolt