boltgolt/howdy #644: dependency updates break howdy, leaves you unable to login.

kanaida2, 2022-04-02:
After updating manjaro, it wanted to update the lib for visual processing, I think by intel. it did.
next i couldn't login anymore and it wasn't taking photos.
you should package it to resolve dependency issues or use your own code for facial recognition.

I can't remember the exact library name but it left linux not accepting passwords. very bad.
I think there's a bug in PAM that doesn't handle crashes in modules very well.
The only account that would login was root.
I had to format my pc. RIP linux.

kanaida2, 2022-04-02:
opencv was the dependency that updated past what howdy supported.

boltgolt, 2022-04-03:
First of all, sorry that happened to you. 

Howdy uses a small script that interfaces between PAM and the core Howdy module, which handles any crashes and does not import OpenCV at all. I'm not sure how you got stuck that badly. Did you see any errors etc? Did you try to disable Howdy from the root prompt you did access?

closed 2026-11-16 by boltgolt