boltgolt/howdy #653: Howdy face detection timeout reached

martinpelli, 2022-04-20:
HI, I really don't know if this is a bug or is a bad configuration that I did, but I searched a lot for days and I can't solve it (or understand what I have to do)

I can't get face recognition working and I have some warnings  when trying to authenticate or when running sudo howdy snapshot.

`[ WARN:0@0.307] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.308] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.308] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Face detection timeout reached
`

What I did to install it was:

yay -S --editmenu howdy    //to install it

mpv /dev/video2       //to test camera IR, Here the camera works it's blink in dark and white and the red sensor turn on.

sudo vim /lib/security/howdy/config.ini           //Here I edited device_path = /dev/video2

sudo howdy add        //Here I get the warnings and  -> Please look straight into the camera /Scan complete/Added a new model to ...

sudo vim /etc/pam.d/sudo      //Here I added pam.py

Linux distribution: Arch LInux

Howdy version 2.6.1

boltgolt, 2022-04-20:
Your camera can not be read, is another program accessing it at the same time?

martinpelli, 2022-04-20:
> Your camera can not be read, is another program accessing it at the same time?

No, I just installed Arch a few days ago, So I don't have many apps, Maybe mpv? Is a video player that I downloaded to test if my IR was on /dev/video2. I will unistall it and reboot and see waht happens

martinpelli, 2022-04-20:
When removed MPV I am getting this:

`[ WARN:0@0.386] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.386] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.386] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
All frames were too dark, please check dark_threshold in config
Average darkness: [86.16612], Threshold: 50.0
Face detection image too dark`

boltgolt, 2022-04-21:
What does a `sudo howdy snapshot` look like?

martinpelli, 2022-04-21:
> What does a `sudo howdy snapshot` look like? (please upload the file)

This was with lights turned off 

`[pelli@PelliArch ~]$ sudo howdy snapshot
[ WARN:0@0.315] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.316] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.316] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Generated snapshot saved as
/usr/lib/security/howdy/snapshots/20220421T101229.jpg`

[Snapshot](https://imgur.com/TMixjdC)

boltgolt, 2022-04-21:
Hm, looks great, please add `export OPENCV_LOG_LEVEL=ERROR` at the bottom of `nano ~/.bashrc` and set `dark_threshold` to 100 in `sudo howdy config`

martinpelli, 2022-04-21:
> Hm, looks great, please add `export OPENCV_LOG_LEVEL=ERROR` at the bottom of `nano ~/.bashrc` and set `dark_threshold` to 100 in `sudo howdy config`

Done. Now getting timeout reached

`[pelli@PelliArch ~]$ sudo howdy snapshot
Face detection timeout reached
[sudo] password for pelli: 
[ WARN:0@0.051] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.051] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.051] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Generated snapshot saved as
/usr/lib/security/howdy/snapshots/20220421T110003.jpg
`

boltgolt, 2022-04-21:
Strange, if you run `sudo howdy test` is there a circle around your face and does the top right say "scan frame" a lot?

martinpelli, 2022-04-21:
> Strange, if you run `sudo howdy test` is there a circle around your face and does the top right say "scan frame" a lot?

Yes there is a circle around my face but It says SCAN FRAME one time, the IR is blinking, same for camera screen

> sudo howdy test
> Identified face as pelli
> [ WARN:0@0.055] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
> [ WARN:0@0.055] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
> [ WARN:0@0.055] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
> 
> Opening a window with a test feed
> 
> Press ctrl+C in this terminal to quit
> Click on the image to enable or disable slow mode
> 
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
> ^C
> Closing window

boltgolt, 2022-04-21:
Well sounds like everything is working perfectly then now, your log even says "Identified face as pelli"

martinpelli, 2022-04-21:
> Well sounds like everything is working perfectly then now, your log even says "Identified face as pelli"

Yes!! but before executing howdy test it wasn't working, Now tried sudo pacman ... and it worked thanks!!

boltgolt, 2022-04-21:
Happy to help!

martinpelli, 2022-04-21:
> Happy to help!

Sorry, but it's possible to add it to gnome? How I cand do it?  Maybe adding to sudo vim /etc/pam.d/sudo but I don't have any ideda of what to put in the line

boltgolt, 2022-04-21:
So [this wiki page](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types) explains part of adding Howdy authentication to specific auth types. It seems that you have added it to sudo specifically. If you add 
```
auth       sufficient   pam_python.so  /lib/security/howdy/pam.py
```

to `/etc/pam.d/common-auth` you should be good to go anywhere. Be sure to remove any lines you've added to pam files before this so Howdy does not get triggered twice

martinpelli, 2022-04-21:
It didn't work in /etc/pam.d/common-auth but I did the same thing in /etc/pam.d/system-auth
and now is working everywere! Again, thanks!!