boltgolt/howdy #655: Howdy crash report on Ubuntu 22.04

cccaballero, 2022-04-24:
Using Ubuntu 22.04 the face recognition works, but I am getting a Howdy related system crash report

[image]

boltgolt, 2022-04-24:
Did recognition succeed? Should not work if you're missing dlib 

Search other issues and you'll find plenty of others with missing dlib dependencies

boltgolt, 2022-04-24:
For example #648

closed 2022-04-24 by boltgolt

cccaballero, 2022-04-25:
I have not had the problem again, I had just updated from 20.04 and reinstalled howdy, it is possible that the error was from before reinstalling howdy and Ubuntu has shown the reports with some delay

Sciroccogti, 2022-05-15:
So a reinstallation is a must, right?

boltgolt, 2022-05-16:
it seems that way