boltgolt/howdy #657: More documentation for rubber stamps

MrConorAE, 2022-04-25:
Hey Howdy team!
First of all, this program works amazingly well. I've been using it alongside open-fprintd and using biometrics to sign in, unlock and authorise is great!

I noticed the "rubber stamps" section in your wiki, but I don't quite understand it. The guide doesn't explain how or where to put the rubber stamp rules (it mentions `stamp_rules`, but it's not clear where & how that should go in the config file).

Could you please explain how to use rubber stamps more thoroughly? I'm happy to edit the wiki if you want to explain it in this issue - I can edit the page myself.

Thanks again for your work :)

boltgolt, 2022-04-26:
Good to hear that Howdy works well for you! Sorry for not being clear about it on the wiki page, but rubber stamps will be introduced in 3.0.0 which is not ready for release yet

closed 2022-04-26 by boltgolt

stdedos, 2024-03-22:
Is there any timeframe for the v3? I see commit activities, but no "some kind of indication" on the timeline.