boltgolt/howdy #660: Sudo authentication does not work in GUI apps

Peitscha, 2022-04-26:
Hey!

I have a problem regarding my installation of Howdy 2.6.1 on Fedora 35. Unfortunately, facial recognition does not work in GUI apps, despite it working on the start up screen and terminal. Specifically, Howdy is caught in an infinite loop and the authentication process only stops after a restart of the whole computer. Just closing the app does not work. Do you know anything that could be the issue in that case?

Thank you very much for your help!

Best regards
Gabriel

boltgolt, 2022-04-26:
Anything related in the auth log? Very concerning that Howdy keeps going after the timeout

Peitscha, 2022-04-27:
I have generated the problem again, and it seems like that there is no timeout. From my understand at looking at the logs, the authentication process always restarts after it has failed. The following log lines repeat endlessly:

`
Attempting facial authentication for user peitscha

[ WARN:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/cap_gstreamer.cpp (2056) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
Unknown line '[ INFO:0] global /builddir/build/BUILD/opencv-4.5.3/modules/videoio/src/videoio_registry.cpp (217) VideoBackendRegistry VIDEOIO: Enabled backends(8, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940); UEYE(930)' from helper

BrokenPipeError: [Errno 32] Broken pipe
MFX: Can't initialize session
libva info: va_openDriver() returns -1
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: VA-API version 1.13.0
`

closed 2026-11-16 by boltgolt