boltgolt/howdy #663: Generate some static fingerprint after a face recognition process is completed

MurzNN, 2022-04-28:
Is it possible to generate some static fingerprint (string) after face recognition process is completed? It can be used to encrypt/decrypt user's password, that makes possible to pass the decrypted password to other modules like`pam_gnome_keyring.so` and `pam_kwallet5.so`, to automatically unlock them by face too.

boltgolt, 2022-04-28:
While not impossible that's going to be incredibly hard to do in a consistent way. Even matching stored (plain text) landmarks on a users face to landmarks extracted from a video frame is an imprecise science. A detected face is _never_ exactly the same shape as stored on disk, and that is why Howdy has a `certainty` config value that is always higher than 1.

What consistent biometric could we extract that always gives the exact same hash that's also long enough not to be cracked?

MurzNN, 2022-04-30:
Thank you for the explanation! I'm wondering how this works on Windows, cuz it's "Hello" engine logs me in into the account without asking me to type password to decrypt the secrets storage, and all stored passwords become accessible without any additional actions...

boltgolt, 2022-04-30:
I'm not entirely sure, worth looking into