boltgolt/howdy #664: screensaver unlock xfce4 recognizing face but not unlocking

seanhinkley81, 2022-04-28:
I am running ZorinOS (Kernel 5.13.0-40/xfce4) I have installed howdy and it works at the login screen no issues, it was not working at the screensaver unlock, I found an answer for kde & another for xfce in and applied chmod a+x  to /lib/security/howdy/models and /lib/security/howdy/recorders and now it is failing successfully?  Instead of an error message i see the message it recognizes me, and then nothing. The password prompt remains.  I have edited pam.d/xfce4-screensaver and pam.d/lightdm  per the install instructions. Overall howdy is working really well and I like it except for this one issue. 

[image]

boltgolt, 2022-04-28:
> failing successfully

Yeah that's a weird case. Maybe disabling the confirmation message in the config will fix it? That sometimes trips up screensavers

seanhinkley81, 2022-04-28:
Changing 
no_confirmation = false  to true
in the config file
Works.  Not something I would have thought of

seanhinkley81, 2022-04-28:
Workaround

Edit Config.ini 
change

# Do not print anything when a face verification succeeds
no_confirmation = _**true**_

closed 2022-04-28 by seanhinkley81

boltgolt, 2022-04-28:
Yeah it's bizarre, good that it fixed it