boltgolt/howdy #666: brandnew MXlinux 21.1 KDE Debian 11 based via .deb with this issue

homa57, 2022-05-01:
$ sudo howdy -U ich add
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 48, in <module>
    face_encoder = dlib.face_recognition_model_v1(path + "/../dlib-data/dlib_face_recognition_resnet_model_v1.dat")
RuntimeError: Unable to open /usr/lib/security/howdy/cli/../dlib-data/dlib_face_recognition_resnet_model_v1.dat for reading.

boltgolt, 2022-05-01:
Does `usr/lib/security/howdy/dlib-data/dlib_face_recognition_resnet_model_v1.dat` exist? What are the file permissions?

homa57, 2022-05-02:
No, Download the howdy_2.6.1.deb file from the [Releases page](https://github.com/boltgolt/howdy/releases) and install with gdebi.

```
root@mx21:~# cd /usr/lib/security/howdy/
root@mx21:/usr/lib/security/howdy# ls
cli  cli.py  compare.py  config.ini  dlib-data  logo.png  pam-config  pam.py  __pycache__  recorders  snapshot.py
root@mx21:/usr/lib/security/howdy# cd dlib-data/
root@mx21:/usr/lib/security/howdy/dlib-data# ls
install.sh  mmod_human_face_detector.dat  Readme.md  shape_predictor_5_face_landmarks.dat
```

boltgolt, 2022-05-02:
Sounds like you had a hiccup with your internet during installation. Run:

```
cd /usr/lib/security/howdy/dlib-data
./install.sh
```

Or purge and reinstall Howdy

closed 2022-05-02 by boltgolt

homa57, 2022-05-02:
Thank you! 

```
/usr/lib/security/howdy/dlib-data# ./install.sh
Downloading 3 required data files...
dlib_face_recognition_resnet_model_v1.dat.bz2   100%[=======================================================================================================>]  20,44M  4,25MB/s    in 6,1s    
mmod_human_face_detector.dat.bz2                100%[=======================================================================================================>] 678,41K  4,21MB/s    in 0,2s    
shape_predictor_5_face_landmarks.dat.bz2        100%[=======================================================================================================>]   5,44M  4,94MB/s    in 1,1s    
Unpacking...
```