boltgolt/howdy #671: Python Exception "AttributeError" on /usr/lib/security/howdy/cli/test.py

NP-chaonay, 2022-05-20:
When using value of gamma that not "-1" in conifguration file given from "howdy config" command

and doing command "howdy test" got this:

```
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 194, in <module>
    video_capture.intenal.set(cv2.CAP_PROP_AUTO_EXPOSURE, 1.0)  # 1 = Manual
AttributeError: 'VideoCapture' object has no attribute 'intenal'. Did you mean: 'internal'?
```

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 22.04 LTS

Howdy version (`sudo howdy version`): Howdy 2.6.1
Howdy from official repo, I have looked through beta/stable/dev branch seem to be no fixed yet.

NP-chaonay, 2022-05-20:
changing intenal to internal as following exception description fixed this issues.

NP-chaonay, 2022-05-20:
changed to pull request instead. (https://github.com/boltgolt/howdy/pull/672)

closed 2022-05-20 by NP-chaonay

boltgolt, 2022-05-24:
Thought we fixed this typo everywhere, thanks for the PR!