boltgolt/howdy #674: Face recognition doesn't work

SaLukasik, 2022-05-26:
_Please describe the issue in as much detail as possible, including any errors and traces._

Hi there! I'm using Ubuntu on Yoga Slim 7. I've configured my IR camera to blink in red light and it seems to work poperly. The problem is howdy is working worse since then (before fixing my ir camera it worked sometimes, rather poorly but now  it doesn't work at all). When I boot to Windows my ir camera works great and detects my face in few sec or faster. I don't get any error messages, just info that my face wasn't detected.
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

----

I've searched for similar issues already, and my issue has not been reported yet.
I've tried:
-Adding a lot of photo samples with and without glasses
-Plaing with configurations, including resolution and other options
-Disabling some lockscreeng safety options as I found in some solutions
Linux distribution (if applicable): 21.10

Howdy version (`sudo howdy version`): Howdy 2.6.1

My current configurations:
# Howdy config file
# Press CTRL + X to save in the nano editor

[core]
# Print that face detection is being attempted
detection_notice = false

# Do not print anything when a face verification succeeds
no_confirmation = true

# When a user without a known face model tries to use this script, don't
# show an error but fail silently
suppress_unknown = false

# Disable Howdy in remote shells
ignore_ssh = true

# Disable Howdy if lid is closed
ignore_closed_lid = true

# Disable howdy in the PAM
# The howdy command will still function
disabled = false

# Use CNN instead of HOG
# CNN model is much more accurate than the HOG based model, but takes much more
# computational power to run, and is meant to be executed on a GPU to attain reasonable speed.
use_cnn = false

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
# Lower is better
certainty = 2.8

# The number of seconds to search before timing out
timeout = 4

# The path of the device to capture frames from
# Should be set automatically by an installer if your distro has one
device_path = /dev/video2

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 480

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = 848
frame_height = 480

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 50

# The recorder to use. Can be either opencv (default), ffmpeg or pyv4l2.
# Switching from the default opencv to ffmpeg can help with grayscale issues.
recording_plugin = opencv

# Video format used by ffmpeg. Options include vfwcap or v4l2.
# FFMPEG only.
device_format = v4l2

# Force the use of Motion JPEG when decoding frames, fixes issues with YUYV
# raw frame decoding.
# OPENCV only.
force_mjpeg = false

# Specify exposure value explicitly. This disables autoexposure.
# Use qv4l2 to determine an appropriate value.
# OPENCV only.
exposure = -1

[snapshots]
# Capture snapshots of failed login attempts and save them to disk with metadata
# Snapshots are saved to the "snapshots" folder
capture_failed = true

# Do the same as the option above but for successful attempts
capture_successful = true

[debug]
# Show a short but detailed diagnostic report in console
# Enabling this can cause some UI apps to fail, only enable it to debug
end_report = false

SaLukasik, 2022-05-27:
> Does sudo howdy test show a circle around your face?

Yes it does very fast.

boltgolt, 2022-05-28:
Then there's nothing wrong with your setup, your face just does not match the saved models. Try clearing the current models with `sudo howdy clear` and adding 1 or 2 new ones. I can also see your `certainty` config value is quite low, try upping that a bit

closed 2026-11-16 by boltgolt