boltgolt/howdy #675: $ sudo howdy add Adding face model for the user wheeler Enter a label for this new model [Initial model] (max 24 characters): asd Camera path is not configured correctly, please edit the 'device_path' config value. Exception ignored in: <function VideoCapture.__del__ at 0x7fe8a873ac20> Traceback (most recent call last):   File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__     self.internal.release() AttributeError: 'VideoCapture' object has no attribute 'internal' wheeler@fedora:~$

wheelerlaw, 2022-06-01:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

```
$ sudo howdy add
Adding face model for the user wheeler
Enter a label for this new model [Initial model] (max 24 characters): asd
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7fe8a873ac20>
Traceback (most recent call last):
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
wheeler@fedora:~$ 
```

Running `sudo howdy snapshot` results in:

```
$ sudo howdy snapshot
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7efd1b313010>
Traceback (most recent call last):
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

MrConorAE, 2022-06-01:
I fixed it in my instance by running `sudo howdy config` to open the config file and changing the video device to `/dev/video0`.

I think this error occurs when the automatic installer can't or doesn't automatically detect your camera, and so the value in the config stays at the (invalid) default.

boltgolt, 2022-06-01:
Please read the output you posted: 
```
Camera path is not configured correctly, please edit the 'device_path' config value.
```

closed 2022-06-01 by boltgolt

wheelerlaw, 2022-06-01:
@boltgolt Please refrain from snarky and unhelpful language. I read the output and it wasn't self evident. 

It would seem that there is a bug in the automatic configuration. At the very least, enhancing the documentation on how to resolve the issue would suffice for an acceptable resolution rather than just dismissing the issue by closing the ticket.