boltgolt/howdy #676: Howdy authentication doesn't work.

aman-upadhyay, 2022-06-02:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

I have installed the latest Howdy version from AUR and there was no problem installing it, I was also able to add faces, and take snapshots and the howdy test also works as expected. But I still don't see any howdy msgs during login or sudo. I am a bit confused here about what to do. 

howdy test shows these warnings
```
[ WARN:0@0.170] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.171] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.171] global /build/opencv/src/opencv-4.5.5/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```
But shows correct image

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): manjaro

Howdy version (`sudo howdy version`):  Howdy 2.6.1

boltgolt, 2022-06-02:
Did you add Howdy to any PAM config files?

aman-upadhyay, 2022-06-02:
Sorry, I fixed it and Howdy works now!
But it still gives me all the warnings when I use sudo, is there a way suppress these warnings?

boltgolt, 2022-06-02:
Checkout [this wiki page](https://github.com/boltgolt/howdy/wiki/Common-issues#opencv-warning) for the first question. As for the second question: There's nothing that can be done about that unfortunately (short of submitting a patch to the maintainers).