boltgolt/howdy #677: I can help with translate

bpavuk, 2022-06-11:
Hi, boltgolt. I like to using Howdy auth. It has sent to the russian warship a couple of problems (I like to use big strange hard passwords). But, I have'nt seen russian or Ukrainian language support. I can help you with translating to those languages. It will make this tool really popular, and we'll make our contribution to war between Linux and Shindows (only native face id and games are things that Linux really need to win).

bpavuk, 2022-06-11:
Write me if you'll need my help boopentiy@gmail.com

boltgolt, 2022-06-12:
You can! Checkout the pinned issue at #486

boltgolt, 2022-06-12:
As of right now, the russian translation is done but the ukrainian needs a lot more help

closed 2022-06-12 by boltgolt

bpavuk, 2022-06-12:
Send me what to translate. I'll send you translated text