boltgolt/howdy #678: Face detection fails in the first ~500 milliseconds, it doesn't pass again even with timeout.

EricAndrechek, 2022-06-14:
When using howdy, it works with great accuracy if I am directly staring at the camera. If I am not looking at the camera, it gives certainties in the mid-4s. I tried bumping up my certainty to in the 5s, but a friend that looks relatively similar to me was able to unlock it.  

My problem is that when I have a certainty value of 2-3 it always marks me as unable to detect. When I look at the snapshots it generates, I am never looking right at the camera. I will try to hit enter on anything requiring sudo and look over at my camera really quickly, which often works. If I am slower than roughly half a second to look over though, it always fails. I have the timeout set to 4 right now, but it doesn't seem to work regardless of what timeout I set.

There are two ways I could see approaching this problem for a fix: 
- A more general solution where I can dig up any logs you may need and try anything you need me to. Hopefully, this may help anyone else experiencing this issue or something similar.
- OR a much more simple approach I believe would work just fine that I was kind of surprised wasn't already there, was a delay option. In the config, I could specify like `delay = 1` to tell it to wait 1 second before looking at my camera. This way I'd have time to turn my head without panicking haha!

Let me know what you think of this little feature request idea, or if you'd like some more information about my issue. 

Thanks!

Linux distribution: Linux Mint (Cinnamon)

Howdy version: 2.6.1 (fresh install from apt repository)

Camera: Logitech C922x Pro Stream Webcam – Full 1080p HD Camera

boltgolt, 2022-06-14:
I personally think the real problem here is the timeout failing to work properly. If Howdy would respect the timeout value then you don't need to react so quickly. A delay in which Howdy does absolutely nothing seems like a waste if Howdy could be trying to identify you instead?

Could you send me exactly what is in your config?

boltgolt, 2022-06-14:
At the very least this bug report has made me spot a bug in the compare code: `timeout`, `dark_threshold`, and `end_report` get assigned twice in the compare script and the second time it gets [assigned without the fallback value](https://github.com/boltgolt/howdy/blob/master/src/compare.py#L155). My theory would be that you have a funky timeout value in your config and because of this bug no safe fallback is used

EricAndrechek, 2022-06-14:
Yes, I agree, I just didn't know if maybe there was something specific with my camera that may be better fixed with a delay or something. Fixing the root problem would be great though!

Here is my config:
```
# Howdy config file
# Press CTRL + X to save in the nano editor

[core]
# Print that face detection is being attempted
detection_notice = false

# Do not print anything when a face verification succeeds
no_confirmation = true

# When a user without a known face model tries to use this script, don't
# show an error but fail silently
suppress_unknown = false

# Disable Howdy in remote shells
ignore_ssh = true

# Disable Howdy if lid is closed
ignore_closed_lid = false

# Disable howdy in the PAM
# The howdy command will still function
disabled = false

# Use CNN instead of HOG
# CNN model is much more accurate than the HOG based model, but takes much more
# computational power to run, and is meant to be executed on a GPU to attain reasonable speed.
use_cnn = true

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
# Lower is better
certainty = 5

# The number of seconds to search before timing out
timeout = 1

# The path of the device to capture frames from
# Should be set automatically by an installer if your distro has one
device_path = /dev/v4l/by-path/pci-0000:0c:00.3-usb-0:3:1.0-video-index0

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = -1
frame_height = -1

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 50

# The recorder to use. Can be either opencv (default), ffmpeg or pyv4l2.
# Switching from the default opencv to ffmpeg can help with grayscale issues.
recording_plugin = opencv

# Video format used by ffmpeg. Options include vfwcap or v4l2.
# FFMPEG only.
device_format = v4l2

# Force the use of Motion JPEG when decoding frames, fixes issues with YUYV
# raw frame decoding.
# OPENCV only.
force_mjpeg = false

# Specify exposure value explicitly. This disables autoexposure.
# Use qv4l2 to determine an appropriate value.
# OPENCV only.
exposure = -1

[snapshots]
# Capture snapshots of failed login attempts and save them to disk with metadata
# Snapshots are saved to the "snapshots" folder
capture_failed = true

# Do the same as the option above but for successful attempts
capture_successful = true

[debug]
# Show a short but detailed diagnostic report in console
# Enabling this can cause some UI apps to fail, only enable it to debug
end_report = false
```
 I currently have it set to 1 since it seems to only use the first half second, so the extra time waiting just means I can't type while I wait for it to finish and fail. I have also set it to higher values like 4 and 5 without success. Let me know if you'd like me to try any other numbers or anything. Thanks!

boltgolt, 2022-06-16:
Your config looks good. Does your IR camera stay on in other programs like cheese? Are there any errors in the console or in /var/log/auth.log?

closed 2026-11-16 by boltgolt