boltgolt/howdy #679: Can not unlock in kde 5.25

xz-dev, 2022-06-20:
cat /etc/pam.d/kde
```
#%PAM-1.0

auth            sufficient      pam_python.so /lib/security/howdy/pam.py

auth            include         system-login

account         include         system-login

password        include         system-login

session         include         system-login

```

----

Linux distribution (if applicable): archlinux

Howdy version (`sudo howdy version`): Howdy 2.6.1

xz-dev, 2022-06-20:
try unlock log
[image]

chinggg, 2022-06-20:
Same issue after updating KDE.
```
auth sufficient pam_unix.so try_first_pass likeauth nullok
auth sufficient pam_python.so /lib/security/howdy/pam.py
```
Adding these two lines in `kde`, `system-local-login`, `system-login` does not work

MurzNN, 2022-06-20:
Same problem, and here is `journalctl -f` logs when the unlock process is failed:
```
Jun 20 09:30:48 UBUNTU [14865]: Attempting facial authentication for user alexey_korepov
Jun 20 09:30:50 UBUNTU audit: MAC_TASK_CONTEXTS subj_apparmor=unconfined
Jun 20 09:30:51 UBUNTU kernel: QThread[14867]: segfault at 8 ip 00007fb39a81ba50 sp 00007fb39b022140 error 4 in pam_python.so[7fb39a818000+5000]
Jun 20 09:30:51 UBUNTU kernel: Code: ff ff 44 89 f6 48 89 df e8 4d e3 ff ff 83 f8 ff 0f 84 9a fe ff ff 0f 1f 40 00 8b 44 24 14 85 c0 75 30 48 8b 44 24 28 48 89 df <8b> 50 08 48 8b 30 e8 b5 e2 ff ff 49 89 c7 e9 73 fe ff ff 0f 1f 44
Jun 20 09:30:50 UBUNTU audit[14865]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=5 subj=? pid=14865 comm="QThread" exe="/usr/lib/x86_64-linux-gnu/libexec/kscreenlocker_greet" sig=11 res=1
Jun 20 09:30:51 UBUNTU ksmserver[14911]: Qt: Session management error: networkIdsList argument is NULL
```

boltgolt, 2022-06-20:
Anything in `/var/log/auth.log`?

Integral-Tech, 2022-06-20:
Same problem after upgrading to KDE Plasma 5.25.
`6月 20 22:19:05 IntegralPC [16632]: Attempting facial authentication for user integral
6月 20 22:19:05 IntegralPC ksmserver[16676]: [ WARN:0@0.384] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
6月 20 22:19:05 IntegralPC ksmserver[16676]: [ WARN:0@0.385] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
6月 20 22:19:05 IntegralPC ksmserver[16676]: [ WARN:0@0.385] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
6月 20 22:19:08 IntegralPC kernel: QThread[16634]: segfault at 8 ip 00007fec3a520180 sp 00007fec3adabff0 error 4 in pam_python.so[7fec3a51d000+5000]
6月 20 22:19:08 IntegralPC kernel: Code: ff ff 44 89 f6 48 89 df e8 cd e2 ff ff 83 f8 ff 0f 84 a3 fe ff ff 0f 1f 40 00 8b 44 24 14 85 c0 75 30 48 8b 44 24 28 48 89 df <8b> 50 08 48 8b 30 e8 35 e2 ff ff 49 89 c7 e9 7c fe ff ff 0f 1f 44`

boltgolt, 2022-06-20:
Could be related to them dropping kcheckpass?

MurzNN, 2022-06-20:
Seems already reported, I've found this https://bugs.kde.org/show_bug.cgi?id=455608

boltgolt, 2022-06-21:
Fix is in on the KDE side, should be fixed on next release

erikatl, 2022-06-21:
Fix already in Manjaro Unstable (KDE 5.25.1). Wow that was quick!

boltgolt, 2022-06-21:
Interesting, it was marked for release 5.25.2

xz-dev, 2022-06-22:
Not fixed in KDE 5.25.1(archlinux)

yurusov, 2022-06-28:
Fixed in KDE 5.25.2, just got released today(arch)
Note that for better experience you should use howdy pam module after pam_unix in /etc/pam.d/kde like this:
```
auth       sufficient                  pam_unix.so          try_first_pass likeauth nullok
auth       sufficient                  pam_python.so /lib/security/howdy/pam.py
```
If you try to run howdy pam before password auth for some reason klocker would execute it 3 times in a row and hangs if failed even once. But I guess it's a different bug.

closed 2022-06-29 by xz-dev

MurzNN, 2022-06-29:
@yurusov Thanks for the example! So it should act only on klocker, without changing behavior of `sudo` and other commands, that now passes by howdy without waiting for entered password?

> But I guess it's a different bug.

So maybe submit it as a separate bug to KDE, for giving a chance to fix it?

yurusov, 2022-06-29:
> @yurusov Thanks for the example! So if I put this into ` /etc/pam.d/kde` - it should act only on `klocker`, without changing behavior of `sudo` and other commands, that now passes by howdy without waiting for entered password?

Yes, it might be used by some other kde apps, not sure which ones, but not sudo, nor sddm, so even if something breaks you still can login.

> So maybe submit it as a separate bug to KDE, for giving a chance to fix it?

I'm not sure if it's KDE bug tbh. And also it never worked this way before, so I kinda used to current behavior. Only difference is extra "enter" key press, which is fine by me.