boltgolt/howdy #680: Howdy fails to install in Ubuntu 22.04, unable to download three data files

SolitaryDragon, 2022-06-22:
I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution: Ubuntu 22.04 LTS

Howdy version (`sudo howdy version`): looks like 2.6.1

At the end of my Howdy installation, the terminal prompts me that it needs to download three data files. Then it tells me that the installation failed "No such file or directory". It looks like a network issue and the files cannot be downloaded. I want to download these three files manually, but I don't know what the three files are and what directory to put them in. Hope to get an answer.

The following is the terminal input information:

~$ sudo apt install howdy 

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
howdy is already the newest version (2.6.1).
The following packages were automatically installed and are no longer required:
  ibus-data ibus-gtk4 libmarisa0 libopencc-data libopencc1.1 libpinyin-data
  libpinyin13 python3-ibus-1.0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up howdy (2.6.1) ...
Attempting installation of missing data files
Downloading 3 required data files...
Unpacking...
bzip2: Can't open input file *.bz2: No such file or directory.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit
 status 1
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

[image]

boltgolt, 2022-06-23:
Have tried to install a second time yet?

boltgolt, 2022-06-23:
Alternatively, you could download them manually:

```
cd /lib/security/howdy/dlib-data
wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
wget https://github.com/davisking/dlib-models/raw/master/mmod_human_face_detector.dat.bz2
wget https://github.com/davisking/dlib-models/raw/master/shape_predictor_5_face_landmarks.dat.bz2
```

SolitaryDragon, 2022-06-23:
> Alternatively, you could download them manually by running this as root:

Thanks for your help, this method works

closed 2022-06-23 by boltgolt

SolitaryDragon, 2022-06-25:
Update: The manual download to the specified path is invalid when the data file cannot be downloaded, the operation cannot be unpacked correctly, and Howdy cannot be used even after the installation is completed.
The correct way should be to solve the network problem and download the data files during the installation process through a proxy, so that they can be unpacked correctly.

boltgolt, 2022-06-25:
The download works for everyone that's not behind strange firewalls like the great firewall. Downloading the files like this in stead of in the .deb is a licensing issue

cityji, 2023-12-25:
I already downloaded above suggested files (as mentioned above) 
and moved to specified location too,

i am trying to install it correctly, as running 
`sudo spkg --configure -a`

_it returns : _

`dpkg --configure -a
Setting up howdy (2.6.1) ...
Attempting installation of missing data files
Downloading 3 required data files...`

**I am totally clueless about what to do now?**
how to install it correctly?