boltgolt/howdy #684: sudo howdy add doesn't work

jkvhjggfhjyhgggh, 2022-07-02:
_Please describe the issue in as much detail as possible, including any errors and traces._

this is what happens when i do sudo howdy add:

Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7fedd94468c0>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f33560288b0>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

that happens when i do sudo howdy snapshot
----

I've searched for similar issues already, and my issue has been reported but the solution isn't applicable to me.

Linux distribution (if applicable): Pop!_OS

Howdy version (`sudo howdy version`):
Howdy 2.6.1

boltgolt, 2022-07-02:
Please read the output you posted: `Camera path is not configured correctly, please edit the 'device_path' config value.`

closed 2022-07-02 by boltgolt

jkvhjggfhjyhgggh, 2022-07-02:
but how do i do that

boltgolt, 2022-07-03:
https://github.com/boltgolt/howdy/wiki/Common-issues#error-when-trying-to-add-a-face-model