boltgolt/howdy #688: Hardcoded installation path

EmixamPP, 2022-07-16:
I was looking at the work that has been done on the new PAM module in C++, and I realized that it would not work on Fedora because the "/lib/security/howdy" path used is not the same.

Then I looked in general in all the files and this problem comes back in others files (I had already noticed it and fixed it for onboarding.py). 

I think something should be done to elegantly access the howdy installation folder, instead of using a hardcoded path, or as I did for onboarding.py, look at which of the possible paths exist. 

I suggest using a global variable that will be set to the correct path during installation is good practice. It would be enough to create a file `/etc/profile.d/howdy.sh` which contains `export HOWDY_LIB=...`.
To get the lib path in C++ we could use `std::getenv("HOWDY_LIB")`, and in Python `os.environ['HOWDY_LIB']`.

This is just a proposal to facilitate the installation of Howdy regardless of the distro or method used. I can create a PR with the appropriate modifications if you wish?  But first I wanted to be sure that this would be accepted.

boltgolt, 2022-07-16:
Sounds good to me! I really want to move Howdy to separate folders too, the current location is just used because of the pam-python dependency

eli-schwartz, 2022-07-22:
> I suggest using a global variable that will be set to the correct path during installation is good practice.

Adapting to the OS paths is a good practice.

Doing this via an environment variable is a terrible practice. :) Compile the correct location into the built files with e.g. Meson's `configure_file()` or compiler macros such as `-DHOWDY_LIB=/path/to/lib`.

closed 2022-07-23 by EmixamPP