boltgolt/howdy #694: Howdy doesn't work on archlinux

rodude123, 2022-07-28:
I've followed the guides to installing howdy on Arch Linux, I've installed and set it up. I've added the pam configs to all the files in `pam.d` directory but it still doesn't work. What have I done wrong because it doesn't show up when I try a command with sudo

ECO1AI, 2022-07-30:
same thing for me 
i am running manjaro 

it seems like aur packages are un-well-maintained due to lack of maintainers 

i will try the git version or compiling from source and i will see if that fixs the problem 

its worth to mention that adding user is done smoothly but then when i login or use sudo howdy does not work.

wmyfelix, 2022-07-31:
Though the AUR lacks maintenance, it still works on my Manjaro. You can also try the howdy-beta-git for the latest, but I guess the version is not the problem but something else. Have you tried `sudo howdy test`? Does it work?

ECO1AI, 2022-07-31:
`sudo howdy test` works
While operating I didn't see any warning nor errors

I suspect that the problem is between howdy and sudo; since howdy seccesfully recognized my face. However the password request doesn't reach howdy. Do howdy have a service in systemd?

As of the beta version, there are dependencies that pacman cannot provide.

wmyfelix, 2022-07-31:
The beta version works here.  

```
❯ yay -Qi howdy-beta-git
Name            : howdy-beta-git
Version         : 2.6.1.r147.g943f1e1-1
Description     : Windows Hello for Linux (Beta)
Architecture    : x86_64
URL             : https://github.com/boltgolt/howdy
Licenses        : MIT
Groups          : None
Provides        : howdy
Depends On      : opencv  hdf5  python  python-pillow  python-dlib  python-face_recognition  python-face_recognition_models  python-click  python-numpy  python-opencv
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : howdy
```
Because some failure will cause `return pamh.PAM_AUTHINFO_UNAVAIL` that shows no error but directly goes to the next pam authentication stage. You can add some debug prints into pam.py,  an example snippet for the latest git version corresponded config, which is different from the AUR:

```py
def doAuth(pamh):
        """Starts authentication in a separate process"""
        #pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Howdy is Starting 0."))
        # Abort if Howdy is disabled
        if config.getboolean("core", "disabled"):
                pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Howdy is disabled."))
                cameraSwitch(pamh, "off")
                return pamh.PAM_AUTHINFO_UNAVAIL
        #pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Howdy is Starting 1."))
        # Abort if we're in a remote SSH env
        if config.getboolean("core", "abort_if_ssh"):
                if "SSH_CONNECTION" in os.environ or "SSH_CLIENT" in os.environ or "SSHD_OPTS" in os.environ:
                        pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Howdy is disabled in ssh."))
                        cameraSwitch(pamh, "off")
                        return pamh.PAM_AUTHINFO_UNAVAIL
        #pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Howdy is Starting 2."))
        # Abort if lid is closed
        if config.getboolean("core", "abort_if_lid_closed"):
                if any("closed" in open(f).read() for f in glob.glob("/proc/acpi/button/lid/*/state")):
                        pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "LID is closed."))
                        cameraSwitch(pamh, "off")
                        return pamh.PAM_AUTHINFO_UNAVAIL
        #pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Howdy is Starting 3."))
        # Abort if the video device does not exist
        if not os.path.exists(config.get("video", "device_path")):
                if config.getboolean("video", "warn_no_device"):
                        pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Camera path is not configured correctly, please edit the 'device_path' config value."))
                cameraSwitch(pamh, "off")
                return pamh.PAM_AUTHINFO_UNAVAIL
        #pamh.conversation(pamh.Message(pamh.PAM_TEXT_INFO, "Howdy is Starting 4."))

```

ECO1AI, 2022-08-01:
it seems like here is the issue 
there is not pam.py file 
i can install [python for pam ](https://aur.archlinux.org/packages/pam-python)or i can install [pam face authentication ](https://aur.archlinux.org/packages/pam-face-authentication)

however pam (official) repository doesn't have pam.py nor howdy beta

Vanadiom, 2023-01-07:
There actually is a pam.py file https://github.com/boltgolt/howdy/blob/beta/howdy/src/pam.py
It should be located unde `/lib/security/howdy/pam.py`

closed 2026-11-16 by boltgolt