boltgolt/howdy #695: Refresh howdy devices

Mavarice, 2022-08-04:
I sometimes switch around my USB devices (unrelated to Howdy) which causes the configuration to become borked and I have to use 
`v4l2-ctl --list-devices` to list the new devices, and then try them until one works.

During initial set up of Howdy it actually cycles through the devices and asks when the IR emitter fires and then sets that in the config. It would be awesome if there was a way to relaunch that process on the fly.
Maybe `sudo howdy config --refresh` or something?

Mavarice, 2022-08-04:
Thank you so much for running Howdy, it makes using my system dramatically more convenient and I am incredibly grateful, regardless on your response to the above request. Thank you.

boltgolt, 2022-08-05:
Could you reply with your configured device_path? There might be a much easier solution

Mavarice, 2022-08-05:
So on a fresh install and going through the "Is the IR emitter running" prompts it chooses:
`device_path = /dev/v4l/by-path/pci-0000:29:00.3-usb-0:2.1.3:1.0-video-index2`

If I swap the USB devices around it no longer works with the following error:
```
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7fbb1a3c2b90>
Traceback (most recent call last):
  File "/usr/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'
```

but running `v4l2-ctl --list-devices` lists out:
```
Logitech BRIO (usb-0000:29:00.3-2.1.4):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
```

So I can switch it to `/dev/video2` and it functions again. If I swap the USB it actually _continues_ to function, assume since it's on a more generic device path. I am running it through a USB hub, which may be affecting it.