boltgolt/howdy #697: Howdly doesn't work at the login

YulBrinner90, 2022-08-09:
Hi,

Howdly works everywhere except at the login (Ubuntu 22.04). It works at the beginning, to be more precise, but immediatly after Gnome demands me an "authentication request". So I have to type the password manually. It works fine after the suspension. I have added the line "auth sufficient pam_python.so /lib/security/howdy/pam.py" at the end of each file in /etc/pam.d. 

Thanks in advance

boltgolt, 2022-08-09:
So the problem here is that Howdy works fine, but you have passwords or other secrets saved in your keyring that are encrypted with your password. Gnome can't decrypt this without you entering your password first, nothing Howdy can do about that in a secure way. See #438

closed 2022-08-09 by boltgolt

YulBrinner90, 2022-08-09:
Thanks. Sorry for the duplication of the post.