boltgolt/howdy #700: stopped working on Ubuntu 22.04

Liemaeu, 2022-08-13:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Since I upgraded from Ubuntu 20.04 to 22.04 I get this error when running a sudo command (face detection doesn't work):

`data files have not been downloaded, please run the following commands:

	cd /lib/security/howdy/dlib-data
	sudo ./install.sh

[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (2075) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (1053) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 224, in <module>
    face_locations = face_detector(gsframe, 1)
TypeError: 'NoneType' object is not callable
Unknown error: 1`

I already ran the commands above (had to sudo -s before since as a normal user I wasn't allowed to cd to the directory) and rebooted, didn't help. Strangely enough the face detection works on login, but not afterwards in the terminal when running a sudo command.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 22.04

Howdy version (`sudo howdy version`): 2.6.1

ryancastro, 2022-08-15:
If it's helpful, I just removed howdy and reinstalled it immediately. It recompiled and everything works again. (Just to get you up and running again).

Liemaeu, 2022-08-16:
> If it's helpful, I just removed howdy and reinstalled it immediately. It recompiled and everything works again. (Just to get you up and running again).

Thank you very much!

Fixed it.

closed 2022-08-16 by Liemaeu