boltgolt/howdy #708: None of the /dev/videoX works.

magtheo, 2022-09-24:
When i try to add a new face i get this error 

Traceback (most recent call last):
  File "/usr/bin/howdy", line 107, in <module>
    import cli.snap
  File "/usr/lib64/security/howdy/cli/snap.py", line 18, in <module>
    video_capture = VideoCapture(config)
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 34, in __init__
    if not os.path.exists(self.config.get("video", "device_path")):
  File "/usr/lib64/python3.10/configparser.py", line 782, in get
    d = self._unify_values(section, vars)
  File "/usr/lib64/python3.10/configparser.py", line 1153, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'video'
Exception ignored in: <function VideoCapture.__del__ at 0x7f05e609f010>
Traceback (most recent call last):
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

so i stied to look up on the web, which led me to try to change the device_path, none of the different device_path s that i tried has worked. iIhave also tried the IR camera emitter: https://github.com/EmixamPP/linux-enable-ir-emitter. I also tried following this site: https://copr.fedorainfracloud.org/coprs/principis/howdy/. and at last i tried following this site, but i didn't quite understand all the steps, so that might be the problem here: https://linuxreviews.org/Howdy/SELinux

Hope this is good info on my problem :) 

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 36

Howdy version (`sudo howdy version`): 2.6.1

boltgolt, 2022-09-27:
Is there any other output in the console? Are you able to open a camera in any other applications?

magtheo, 2022-10-29:
Hey man, in cheese it says "no device found" but when i open guvcview it shows my webcam perfectly fine. I have no clue what do do in this situation.Thanks for your reply

boltgolt, 2022-10-31:
Seems like a pretty funky setup then, the rule of thumb is that if cheese does not support it then Howdy doesn't either. Maybe you could try changing the recorder to something else than opencv in the config?

magtheo, 2022-11-07:
Thanks for the response, i will come back if i have an update!

closed 2026-11-16 by boltgolt