boltgolt/howdy #711: Nothing is happening on Fedora 36 and Gnome

RenardFute, 2022-10-12:
There is no error shown, because it's seams like nothing is running.
I well added my face, howdy test works but when I do sudo or on the lock screen nothing is happening.

I've searched for similar issues and already gave all permissions to howdy files (777), restarted my computer and reinstalling everything.

Linux distribution : Fedora 36

Howdy version : 2.6.1

RenardFute, 2022-10-13:
There is no mention of howdy in my `journalctl SYSLOG_FACILITY=10`

hys0star, 2022-10-23:
I have the same problem.

yi-ge, 2022-10-28:
I have the same problem.

boltgolt, 2022-10-31:
@RenardFute @hys0star @yi-ge anything in `tail /var/log/auth.log`?

RenardFute, 2022-10-31:
[image]
There is no auth.log on fedora that's why I said `journalctl SYSLOG_FACILITY=10` was empty.

yi-ge, 2022-11-01:
@boltgolt The problem has been resolved. python3 environment config caused the error. 

More: <https://www.wyr.me/post/702>

m1nicrusher, 2022-11-03:
> @boltgolt The problem has been resolved. python3 environment config caused the error.
> 
> More: https://www.wyr.me/post/702

I've tried the python3 solution but have got no luck.
Is it possible it's something to do with SELinux?

m1nicrusher, 2022-11-08:
I've solved the issue on my Surface Laptop Studio.
To be clear, I was always able to create a face model and was able to `sudo howdy test`.
Basically I just followed this link: https://copr.fedorainfracloud.org/coprs/principis/howdy/

In summary, this is what I did:
## Give execute permission to everyone (not sure if needed)
`sudo chmod -R o+x /lib64/security/howdy`
## Configure `sudo` to use Howdy
Open `/etc/pam.d/sudo` and add these to the **top**:
```
auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
``` 
## Configure gdm (if you're using GNOME) to use Howdy to login
Open `/etc/pam.d/gdm-password` and add these to the **second line** (after the first line):
```
auth        sufficient    pam_python.so /lib64/security/howdy/pam.py
```
## Configure SELinux to allow Howdy to access cameras when logging in
(To me, otherwise it will show "unknown error 1" at login)
Create a file at `~` named `howdy.te` and add the following codes:
```
module howdy 1.0;

require {
    type lib_t;
    type xdm_t;
    type v4l_device_t;
    type sysctl_vm_t;
    class chr_file map;
    class file { create getattr open read write };
    class dir add_name;
}

#============= xdm_t ==============
allow xdm_t lib_t:dir add_name;
allow xdm_t lib_t:file { create write };
allow xdm_t sysctl_vm_t:file { getattr open read };
allow xdm_t v4l_device_t:chr_file map;
```
- Then execute the following command:
```
checkmodule -M -m -o howdy.mod howdy.te
semodule_package -o howdy.pp -m howdy.mod
semodule -i howdy.pp
```

ChrisErnovsky, 2022-11-08:
@m1nicrusher, HUGE thanks for this solution!
Works like a charm (Fedora 37 beta, Gnome 43, Lenovo IdeaPad 5 Pro camera + IR).

m1nicrusher, 2022-11-09:
For the convenience of anyone in the future, I made a bash script for Fedora + GNOME.
https://gist.github.com/m1nicrusher/35e79b20553c8863e0c642f8d801da7f
Feel free to point out any error as I have minimal experience working with bash scripts.

Stephen

RenardFute, 2022-11-09:
> For the convenience of anyone in the future, I made a bash script for Fedora + GNOME. https://gist.github.com/m1nicrusher/35e79b20553c8863e0c642f8d801da7f Feel free to point out any error as I have minimal experience working with bash scripts.
> 
> Stephen

Maybe a pr would be helpful for futures users

nuevoparadigma, 2022-11-11:
As @boltgolt writes, the problem has been solved. The cause of the error is that Howdy is running with a version of python 2.
The solution is collected from the following page:

https://www.wyr.me/post/702

In Howdy's pam.py file we find the following code:

# Import required modules
import subprocess
import os
import glob
import syslog

# pam-python is running python 2, so we use the old module here
import ConfigParser

Replace the above with the following and Howdy will work fine:

# Import required modules
import subprocess
import os
import glob
import sys
import syslog

# pam-python is running python 2, so we use the old module here
if sys.version_info.major < 3:
    import ConfigParser
    config = ConfigParser.ConfigParser()
else:
    import configparser
    config = configparser.ConfigParser()

# Read config from disk
config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")

m1nicrusher, 2022-11-11:
@nuevoparadigma Hey mate, you didn't read our conversation here. The problem was not because of Python, but because it was not properly configured with the system. I would assume the reason why other distros work fine is because on Fedora it uses `/lib64/security/howdy` rather than `lib/security/howdy`, plus Fedora has SELinux. I didn't read the code, but maybe there is a script configuring `/lib64/security/howdy` for most distros and it won't work with Fedora for the reason I talked about.

Actually, I'm not sure about the Python thing. I tried replacing the code of `pam.py` but it instead stopped working at all. To anyone who may concern, I didn't change pam.py but only applied my previous solution, and it works fine with Fedora 36 with GNOME.

RedBoardDev, 2022-12-22:
I have the same problem but nothing work

m1nicrusher, 2022-12-23:
> I have the same problem but nothing work

Have you tried the above script? 👇

> https://gist.github.com/m1nicrusher/35e79b20553c8863e0c642f8d801da7f

RedBoardDev, 2022-12-23:
> > I have the same problem but nothing work
> 
> Have you tried the above script? point_down
> 
> > https://gist.github.com/m1nicrusher/35e79b20553c8863e0c642f8d801da7f

I have try and I have the same. Nothing working and my webcam is off when I want to unlock

RenardFute, 2022-12-23:
What does it do when you try ```howdy test```

RedBoardDev, 2022-12-23:
> What does it do when you try `howdy test`

It shows me my face and what the webcam sees. The webcam flashes red, and the iameg on the webcam also flashes. She is black and white

m1nicrusher, 2022-12-24:
> > What does it do when you try `howdy test`
> 
> It shows me my face and what the webcam sees. The webcam flashes red, and the iameg on the webcam also flashes. She is black and white

Seems like the webcam is configured correctly and working fine.

RedBoardDev, 2022-12-24:
Yes, so I don't know where is the problem

RenardFute, 2022-12-24:
> Yes, so I don't know where is the problem. My webcam is not active when I want to unlock my computer. But when I try howdy test, I see the light

It might come from a permission issue check that there is at least  execute permissions for every one. 
I personally put all the howdy directory as 777 but I think it's not recommended 🤷‍♂️

maxjamchuk, 2023-06-01:
I've tried all ways above, however it still not working.

`howdy test` shows my IR camera, but `sudo -i `still asks password.

I noticed that I just don't have `/usr/lib64/security/pam_python.so` file.

PS
`journalctl SYSLOG_FACILITY=10` shows:

```
polkitd[824]: Loading rules from directory /etc/polkit-1/rules.d
polkitd[824]: Loading rules from directory /usr/share/polkit-1/rules.d
polkitd[824]: Finished loading, compiling and executing 15 rules
polkitd[824]: Acquired the name org.freedesktop.PolicyKit1 on the system bus
(systemd)[1083]: pam_systemd_home(systemd-user:account): systemd-homed is not available: Could not activate remote peer: activation request failed: unknown unit.
(systemd)[1083]: pam_unix(systemd-user:session): session opened for user gdm(uid=42) by (uid=0)
gdm-launch-environment][1051]: pam_unix(gdm-launch-environment:session): session opened for user gdm(uid=42) by (uid=0)
polkitd[824]: Registered Authentication Agent for unix-session:c1 (system bus name :1.35 [/usr/bin/gnome-shell], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8)
gdm-password][1894]: PAM unable to dlopen(/usr/lib64/security/pam_python.so): /usr/lib64/security/pam_python.so: cannot open shared object file: No such file or directory
gdm-password][1894]: PAM adding faulty module: /usr/lib64/security/pam_python.so
gdm-password][1894]: gkr-pam: unable to locate daemon control file
gdm-password][1894]: gkr-pam: stashed password to try later in open session
(systemd)[1919]: pam_systemd_home(systemd-user:account): systemd-homed is not available: Could not activate remote peer: activation request failed: unknown unit.
(systemd)[1919]: pam_unix(systemd-user:session): session opened for user maks(uid=1000) by (uid=0)
gdm-password][1894]: pam_unix(gdm-password:session): session opened for user maks(uid=1000) by (uid=0)
gdm-password][1894]: gkr-pam: gnome-keyring-daemon started properly and unlocked keyring
polkitd[824]: Registered Authentication Agent for unix-session:2 (system bus name :1.89 [/usr/bin/gnome-shell], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8)
gdm-launch-environment][1051]: pam_unix(gdm-launch-environment:session): session closed for user gdm
polkitd[824]: Unregistered Authentication Agent for unix-session:c1 (system bus name :1.35, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected from bus)
sudo[5691]: PAM unable to dlopen(/usr/lib64/security/pam_python.so): /usr/lib64/security/pam_python.so: cannot open shared object file: No such file or directory
sudo[5691]: PAM adding faulty module: /usr/lib64/security/pam_python.so
sudo[5691]:     maks : PWD=/root ; USER=root ; COMMAND=/bin/bash
sudo[5691]: pam_unix(sudo-i:session): session opened for user root(uid=0) by (uid=1000)
sudo[5691]: pam_unix(sudo-i:session): session closed for user root
```

stephanlamoureux, 2023-07-02:
Is there a workaround for KDE? I used the bash script previously for GNOME but now I can't find anything for Plasma.

cachapa, 2023-07-27:
I tried all configurations described here and managed to get things mostly working:
- Executing `sudo` on a terminal works
- 1st login after rebooting works (it still asks to unlock the keyring after logging in, but that's ok)

However I can't unlock the screen after locking it. The camera never turns on and "Unknown error: 1" appear below the password field.

It also doesn't seem to have any effect when unlocking Settings (i.e. Users, Printers). The camera doesn't turn on and nothing is shown in the UI.

This is Fedora 38 / Gnome 44.3

1040170566, 2024-05-14:
> To everyone using Fedora 36 and GNOME, I've solved the issue on my Surface Laptop Studio. To be clear, I was always able to create a face model and was able to `sudo howdy test`. Basically I just followed this link: https://copr.fedorainfracloud.org/coprs/principis/howdy/
> 
> In summary, this is what I did:
> ## Give execute permission to everyone (not sure if needed)
> 
> `sudo chmod -R o+x /lib64/security/howdy`
> ## Configure `sudo` to use Howdy
> 
> Open `/etc/pam.d/sudo` and add these to the **top**:
> 
> ```
> auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
> ```
> 
> ## Configure gdm (if you're using GNOME) to use Howdy to login
> 
> Open `/etc/pam.d/gdm-password` and add these to the **second line** (after the first line):
> 
> ```
> auth        sufficient    pam_python.so /lib64/security/howdy/pam.py
> ```
> 
> ## Configure SELinux to allow Howdy to access cameras when logging in
> 
> (To me, otherwise it will show "unknown error 1" at login) Create a file at `~` named `howdy.te` and add the following codes:
> 
> ```
> module howdy 1.0;
> 
> require {
>     type lib_t;
>     type xdm_t;
>     type v4l_device_t;
>     type sysctl_vm_t;
>     class chr_file map;
>     class file { create getattr open read write };
>     class dir add_name;
> }
> 
> #============= xdm_t ==============
> allow xdm_t lib_t:dir add_name;
> allow xdm_t lib_t:file { create write };
> allow xdm_t sysctl_vm_t:file { getattr open read };
> allow xdm_t v4l_device_t:chr_file map;
> ```
> 
>     * Then execute the following command:
> 
> 
> ```
> checkmodule -M -m -o howdy.mod howdy.te
> semodule_package -o howdy.pp -m howdy.mod
> semodule -i howdy.pp
> ```
> 
> Hopefully this helps. Stephen

Verrrrrrrry good! I tried so many methods but only yours works. Thank you! Mine is Ubuntu22.04.4 and Gnome.

Muhd-Mairaj, 2024-07-18:
> Is there a workaround for KDE? I used the bash script previously for GNOME but now I can't find anything for Plasma.

I got it to work on kde by changing the `GDM_CFG="/etc/pam.d/gdm-password"` in the script to `GDM_CFG="/etc/pam.d/kde"`

closed 2026-11-16 by boltgolt