boltgolt/howdy #715: Support for Ubuntu 22.10 (Kinetic Kudu)

mikecharles, 2022-10-21:
I know this just came out, but I wanted to make a placeholder. Is it possible to install this properly on 22.10, despite the fact that there's no repo for it yet?

Unsurprisingly, this is the result of trying to add the apt repo:

```
$ sudo add-apt-repository ppa:boltgolt/howdy
...
Ign:6 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu kinetic InRelease    
Err:7 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu kinetic Release
  404  Not Found [IP: 185.125.190.52 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu kinetic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

roneydua, 2022-10-24:
Hello, 
I downloading from:
https://github.com/boltgolt/howdy/releases/tag/v2.6.1
and works with normal instalation

boltgolt, 2022-10-31:
Thanks for reporting, i've published a new package for kinetic and lunar too while i was at it

closed 2022-10-31 by boltgolt