boltgolt/howdy #721: Create an actual PAM module

LorenDB, 2022-11-04:
Currently, Howdy only supports PAM via pam-python. However, that means that I as an openSUSE user cannot use [`pam-config`](https://documentation.suse.com/sles/15-SP3/html/SLES-all/cha-pam.html#sec-pam-pam-config) to manage adding Howdy to my PAM configuration. This is a slight problem, as pam-config is the intended method of adding PAM modules to your configuration on SUSE-based distros.

I propose that a stub PAM module is created that simply acts as a wrapper for the call to pam-python but is an actual PAM module that can be detected and used by pam-config.

calvinweb, 2022-11-04:
the function of the `pam.py` is quite simple, is just call`compare.py` and check its return value, so i think write it in C isn' a difficult work,

boltgolt, 2022-11-04:
Take a look at the beta branch, this exact thing has already been implemented but I'm unfortunately really really slow in getting it tested and released. There are a few issues with the asynchronous password input and face recognition

closed 2022-11-04 by boltgolt

LorenDB, 2022-11-04:
Oh nice! I can probably help with testing if needed. Let me know if there's anything that does need tested.