boltgolt/howdy #723: Low Light or bad lighting no face recognition

BigCityCat, 2022-11-12:
This is an awesome program and really appreciate the work put in. Using Dell XPS with Ubuntu and Windows. I notice that windows facial unlock is turning on a light with the webcam to assist in low light or bad lighting situations to assist the camera. Is that possible with this application or is the feature not possible. That is really holding this application back. It really only works in perfect lighting conditions.

Well and also it doesn't open the key ring on login so you still have to type in your password. I know that the gnome devs won't allow it due to security reasons.

boltgolt, 2022-11-12:
That works for almost everyone yes, are you sure you are using your IR camera and not the normal visible light webcam?

BigCityCat, 2022-11-13:
Sorry I guess I was on the wrong camera. I changed it and it looks a lot more clear when I test it. It was on video2 but video0 is the correct camera.

closed 2022-11-13 by BigCityCat