boltgolt/howdy #726: CUDA / GPU support not workin

sigmabeta, 2022-11-19:
Howdy's set up and works great for me. However, digging around in the settings I note that there's an option to use the CNN algorithm, which is intended for use with CUDA. Enabling it does not result in GPU usage reported by `nvidia-smi`. 

Digging around the issue tracker, I realized the issue is that `dlib` does not have CUDA support enabled, and that some users were able to fix this by simply using `pip3` to reinstall that library. 

That makes sense; however, `pip3` does this by attempting to build from source. That's great, but it's failing on my machine because of missing CUDA development libraries. The key snippet from the error output:

```
      -- Found BLAS and LAPACK via pkg-config
      CUDA_TOOLKIT_ROOT_DIR not found or specified
      -- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (Required is at least version "7.5")
      -- Found CUDA, but CMake was unable to find the cuBLAS libraries that should be part of every basic CUDA install. Your CUDA install is somehow broken or incomplete. Since cuBLAS is required for dlib to use CUDA we won't use CUDA.
      -- DID NOT FIND CUDA
      -- Disabling CUDA support for dlib.  DLIB WILL NOT USE CUDA
```
Fedora DNF on the other hand, disagrees:

```
Package xorg-x11-drv-nvidia-cuda-libs-3:520.56.06-1.fc37.x86_64 is already installed.
Package xorg-x11-drv-nvidia-cuda-libs-3:520.56.06-1.fc37.i686 is already installed.
```

If I'm misunderstanding and this package does not contain the correct files, please correct me.

Linux distribution (if applicable): Fedora 37
Howdy version: 2.6.1

sigmabeta, 2022-11-19:
Full console output: 
```
pip3 install dlib 

Defaulting to user installation because normal site-packages is not writeable
Collecting dlib
  Using cached dlib-19.24.0.tar.gz (3.2 MB)
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for dlib, since package 'wheel' is not installed.
Installing collected packages: dlib
  Running setup.py install for dlib ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for dlib did not run successfully.
  │ exit code: 1
  ╰─> [1635 lines of output]
      running install
      /usr/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      package init file 'tools/python/dlib/__init__.py' not found (or not a regular file)
      running build_ext
      Building extension for Python 3.11.0 (main, Oct 24 2022, 00:00:00) [GCC 12.2.1 20220819 (Red Hat 12.2.1-2)]
      Invoking CMake setup: 'cmake /tmp/pip-install-fmyee_bm/dlib_6920fc3ef79b449a83b0d2170087c472/tools/python -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-fmyee_bm/dlib_6920fc3ef79b449a83b0d2170087c472/build/lib.linux-x86_64-cpython-311 -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release'
      -- The C compiler identification is GNU 12.2.1
      -- The CXX compiler identification is GNU 12.2.1
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/lib64/ccache/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /usr/lib64/ccache/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Found PythonInterp: /usr/bin/python3 (found version "3.11")
      -- Found PythonLibs: /usr/lib64/libpython3.11.so
      -- Performing Test HAS_CPP14_FLAG
      -- Performing Test HAS_CPP14_FLAG - Success
      -- pybind11 v2.2.4
      -- Using CMake version: 3.24.2
      -- Compiling dlib version: 19.24.0
      -- SSE4 instructions can be executed by the host processor.
      -- AVX instructions can be executed by the host processor.
      -- Enabling AVX instructions
      -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
      -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
      -- Found Threads: TRUE
      -- Found X11: /usr/include
      -- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so
      -- Looking for XOpenDisplay in /usr/lib64/libX11.so;/usr/lib64/libXext.so - found
      -- Looking for gethostbyname
      -- Looking for gethostbyname - found
      -- Looking for connect
      -- Looking for connect - found
      -- Looking for remove
      -- Looking for remove - found
      -- Looking for shmat
      -- Looking for shmat - found
      -- Found system copy of libpng: /usr/lib64/libpng.so;/usr/lib64/libz.so
      -- Found system copy of libjpeg: /usr/lib64/libjpeg.so
      -- Found WebP: /usr/lib64/libwebp.so
      -- Searching for BLAS and LAPACK
      -- Searching for BLAS and LAPACK
      -- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.0")
      -- Checking for module 'cblas'
      --   Found cblas, version 3.10.1
      -- Checking for module 'lapack'
[... 1567 lines cut ...]
          self.run_command(cmd_name)
        File "/usr/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 317, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3.11/site-packages/setuptools/dist.py", line 1229, in run_command
          super().run_command(command)
        File "/usr/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
          cmd_obj.run()
        File "/tmp/pip-install-fmyee_bm/dlib_6920fc3ef79b449a83b0d2170087c472/setup.py", line 134, in run
          self.build_extension(ext)
        File "/tmp/pip-install-fmyee_bm/dlib_6920fc3ef79b449a83b0d2170087c472/setup.py", line 174, in build_extension
          subprocess.check_call(cmake_build, cwd=build_folder)
        File "/usr/lib64/python3.11/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j16']' returned non-zero exit status 2.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> dlib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

perlun, 2023-04-20:
For reference, I am seeing similar compilation errors with `dlib` on Debian bookworm. Python 3.11.

closed 2026-11-16 by boltgolt