boltgolt/howdy #727: Auto-detect dark threshold

MatthewScholefield, 2022-11-21:
Often as in #418 the dark threshold is incorrect. Printing the darkness values for my camera it seems very trivial to detect a reasonable threshold. Would a contribution be accepted that automatically configures this value if it can be detected?

Here are the darkness values for my camera where you can see the dark frames as consistently `>= 99.1`:
```
[82.9427]
[80.70182]
[80.00607]
[99.18099]
[80.11067]
[99.19748]
[80.238716]
[99.23134]
[80.18143]
[99.235245]
[80.00174]
[99.275604]
[79.90408]
[99.26085]
[79.7895]
[99.283424]
[79.90191]
[99.29383]
[79.90885]
[99.29037]
[79.99002]
[99.28993]
[80.08941]
[99.2934]
[80.0395]
[99.28386]
[80.16363]
[99.306854]
[80.44922]
[99.323784]
[80.54948]
[99.3368]
[80.60547]
[99.33507]
[80.609375]
[99.32422]
[80.64192]
[99.33594]
[80.72048]
[99.34201]
[80.932724]
[99.3776]
[80.99089]
[99.390625]
[81.04297]
[99.40972]
[81.08637]
[99.43837]
[81.118484]
[99.442276]
[81.131516]
[99.42751]
[81.35113]
[99.407555]
[81.5217]
[99.38759]
[81.73394]
[99.35156]
[81.86154]
```

boltgolt, 2022-11-21:
I would definitely accept the contribution but i'm not sure it's "very trivial" to set the threshold automatically. 

Some cameras have 2 light frames and 1 dark, some have 1 light frame and 2 dark. Both of our cameras alternate. Camera exposure can vary greatly, what might be a good threshold in one setting might not be the right one in another. What if the user is lit by a non-emitter light source when they run the command? What happens if someone runs this on a visible light camera accidentally?

MatthewScholefield, 2022-11-21:
Since the IR emitter should emit at a relatively constant brightness and dark frames should be relatively constant if you plot the points it most cases there should be two very distinct clusters. We can just do something simple to identify the darker cluster.

More specifically, one idea is we can just split the points into two groups based on the midpoint between the 10th and 90th percentile darkness value. Then if the darker group has a very low range and standard deviation relative to the midpoint we can be pretty confident in the dark threshold being accurate. In the case someone runs it on a regular camera the range of the dark points relative to the midpoint would be very large so we would know to skip setting the dark threshold automatically.

Do you happen to have a flashing IR camera? If so, could you share your darkness values so I can have at least one other sample to test on before submitting a PR?