boltgolt/howdy #728: No ways to install on Ububntu 22.04

Gelliantida, 2022-11-23:
It is no way to install it on Ubuntu 22.04! Not from the repository, not from the deb package. My logs on Russian? But I think you will understand it. This error from the repository installation 
`Игн:16 https://ppa.launchpadcontent.net/tualatrix/ppa/ubuntu jammy InRelease
Ошб:17 https://ppa.launchpadcontent.net/tualatrix/ppa/ubuntu jammy Release
  404  Not Found [IP: 185.125.190.52 443]
Чтение списков пакетов… Готово            
W: https://download.docker.com/linux/ubuntu/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
W: https://dl.winehq.org/wine-builds/ubuntu/dists/focal/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
E: Репозиторий «https://ppa.launchpadcontent.net/tualatrix/ppa/ubuntu jammy Release» не содержит файла Release.
N: Обновление из этого репозитория нельзя выполнить безопасным способом, поэтому по умолчанию он отключён.
N: Информацию о создании репозитория и настройках пользователя смотрите в справочной странице apt-secure(8).
`
And this is from .deb package
`You can always change this setting in the config.
What profile would you like to use? [f/b/s]: 

Распаковывается howdy (2.6.1) …
dpkg: зависимости пакетов не позволяют настроить пакет howdy:
 howdy зависит от libpam-python, однако:
  Пакет libpam-python не установлен.
 howdy зависит от libopencv-dev, однако:
  Пакет libopencv-dev не установлен.
 howdy зависит от cmake, однако:
  Пакет cmake не установлен.

dpkg: ошибка при обработке пакета howdy (--install):
 проблемы зависимостей — оставляем не настроенным
Обрабатываются триггеры для man-db (2.10.2-1) …
При обработке следующих пакетов произошли ошибки:
 howdy
`
It's about requirements. Do you have more ways to install it&)))

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (Ubuntu 22.04):

Howdy version (`2.6.1`):

Gelliantida, 2022-11-23:
And I tried to install it on the two machines with the same OS but different kernels.

boltgolt, 2022-11-24:
First one is an issue with your installation of [this unrelated PPA](https://launchpad.net/~tualatrix/+archive/ubuntu/ppa):
```
E: The "https://ppa.launchpadcontent.net/tualatrix/ppa/ubuntu jammy Release" repository does not contain a Release file.
```

For the second one, try using apt to install howdy and not dpkg

Gelliantida, 2022-11-25:
The first option, with the addition of another PPA, did not work either. But the installation without using dpkg - works! Thank you, the issue is **resolved!**
From the directory of the deb package: **sudo apt install ./howdy_2.6.1.deb**, or use the full path to the deb package.

closed 2022-11-25 by Gelliantida

boltgolt, 2022-11-25:
I did not mean you should install that PPA but that it is already installed a giving apt issues