boltgolt/howdy #729: ASUS ROG G14 (2022) Radeon 6000 authentication issues with camera + IR

sayantandas, 2022-11-25:
Hi there, 
I am facing issue when trying to authenticate using Camera + IR. The authentication is not working all the time. I have to try more than 4-5 times to get a success. 
When I use only the camera , authentication is much faster and hits close to 100% success rate.
My setup

```
 v4l2-ctl --list-devices
USB2.0 HD UVC WebCam: USB2.0 HD (usb-0000:07:00.4-3):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media0
	/dev/media1
```
OS
```
cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.1 LTS"
```
Kernel
```
uname -a
Linux sayantan-ROG-14 6.0.0-1006-oem #6-Ubuntu SMP PREEMPT_DYNAMIC Tue Oct 18 20:01:16 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```
**howdy config**
```
# Use CNN instead of HOG
use_cnn = false

[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
# Lower is better
certainty = 2

device_path = /dev/video2

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 90
```
**Troubeshooting steps** 

I have tried linux-ir-emmitter
```
 sudo linux-enable-ir-emitter configure
INFO: Ensure to not use the camera during the execution.
INFO: Warning to do not kill the process !
Is the ir emitter flashing (not just turn on) ? Yes/No ? Yes
ERROR: Your emiter is already working, skipping the configuration.
ERROR: The configuration has failed.
INFO: Do not hesitate to visit the GitHub ! https://github.com/EmixamPP/linux-enable-ir-emitter/wiki
```

I have tried the following troubleshooting steps from arch wiki https://wiki.archlinux.org/title/Howdy#Testing_your_IR_camera

The camera at /dev/video0 produces 8 images which I can view using an image viewer
```
gst-launch-1.0 v4l2src device=/dev/video0 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
Got EOS from element "pipeline0".
Execution ended after 0:00:00.582118200
Setting pipeline to NULL ...
Freeing pipeline ...

```
However, it fails to produce anything in /dev/video2
```
 gst-launch-1.0 v4l2src device=/dev/video2 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000526251
Setting pipeline to NULL ...
Freeing pipeline ...
```

Please help me toincrase the success rate for camera + IR

Gelliantida, 2022-11-25:
Do you use /dev/video2 path?

Gelliantida, 2022-11-25:
`[video]
# The certainty of the detected face belonging to the user of the account
# On a scale from 1 to 10, values above 5 are not recommended
# Lower is better
certainty = 3.5

# The number of seconds to search before timing out
timeout = 4

# The path of the device to capture frames from
# Should be set automatically by an installer if your distro has one
device_path = /dev/video2

# Scale down the video feed to this maximum height
# Speeds up face recognition but can make it less precise
max_height = 320

# Set the camera input profile to this width and height
# The largest profile will be used if set to -1
# Automatically ignored if not a valid profile
frame_width = -1
frame_height = -1

# Because of flashing IR emitters, some frames can be completely unlit
# Skip the frame if the lowest 1/8 of the histogram is above this percentage
# of the total
# The lower this setting is, the more dark frames are ignored
dark_threshold = 80

# The recorder to use. Can be either opencv (default), ffmpeg or pyv4l2.
# Switching from the default opencv to ffmpeg can help with grayscale issues.
recording_plugin = opencv

# Video format used by ffmpeg. Options include vfwcap or v4l2.
# FFMPEG only.
device_format = v4l2

` 
It's my Asus G14 2022 howdy config. Just change **device_path = /dev/video2** and **dark_threshold = 80**

Gelliantida, 2022-11-25:
[image]

Gelliantida, 2022-11-25:
It's my Asus Rog 14 2022 howdy config settings. Try it.
[image]

Gelliantida, 2022-11-25:
But my kernel 6.0.9-060009-generic

sayantandas, 2022-11-26:
> But my kernel 6.0.9-060009-generic, and it works well

I used the signed kernel from jammy-proposed as I did not remove the secure boot. Your setting seems to work fine. I was able to get it working in most scenarios. Thank you.

Vanadiom, 2023-01-07:
I believe, that the device you want to use is actually `/dev/video1`
At Least it is on my G14.