boltgolt/howdy #731: `Illegal Instruction` when attempting to use howdy

Nos78, 2022-12-03:
I have installed howdy on my Manjaro installation:
* (Lenovo s21e-20 : 2 core Intel Celeron N2840 @ 2.16GHz 2GB RAM).
* Manjaro (Kernel: 5.15.78-1-MANJARO 64bit); KDE Plasma (5.26.3 - Frameworks 5.99.0 - Qt 5.15.7 - Gfx Platform X11)

I installed it (and have also removed and reinstalled it) with `pamac install howdy`, which installed dependancies, and built and installed howdy via AUR.

I also installed Howdy on my Zorin 16 core (ubuntu based) laptop, and dispite a couple of teething issues (failing to find the binary after installation) a remove and re-install fixed this. I noted that on the second install, the auto-config of my camera worked whereas during the first build/install, this auto-config completely failed and I had to manually configure it - which turned out to be pointless without the binary existing at /usr/local/bin/howdy :smile: This installation of howdy now works seemlessly, but I cannot quite figure out what the difference is between this and my Manjaro version - save for differing distros and differing hardware.

**Issue**
I'm getting this `Illegal instruction` when attempting to run the  `sudo howdy add` command, 

```shell
    [noscere@meg ~]$ sudo howdy add
    [sudo] password for noscere: 
    Illegal instruction
```

**Additional information**
As requested, here is the output from `sudo howdy snapshot`
```shell
    [noscere@meg ~]$ sudo howdy snapshot
    [ WARN:0@0.616] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | 
    GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
    [ WARN:0@0.619] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | 
    GStreamer warning: unable to start pipeline
    [ WARN:0@0.619] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | 
    GStreamer warning: GStreamer: pipeline have not been created
    Generated snapshot saved as
    /usr/lib/security/howdy/snapshots/20221203T005538.jpg
```
I'm not going to post the picture (I don't like mugshots of me :laughing: ), instead I shall describe it - there are four snapshots of my face in a row, with a black bar beneath them, on the right is a "HOWDY for linux" watermark and on the left the information bar says:
```
    GENERATED SNAPSHOT
    Date 2022/12/03 00:55:38 UTC
    Dark threshold config: 50.0
    Certainty config: 3.5
```

As suggested in another thread reporting a similar issue from 2018 [https://github.com/boltgolt/howdy/issues/101](https://github.com/boltgolt/howdy/issues/101) (issue now closed, reporting `Illegal instruction` on a debian-based install of howdy), here is the output from /proc/cpuinfo

```bash
    [noscere@meg ~]$ grep -Em1 --color=always 'sse|avx' /proc/cpuinfo
    flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx 
    fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology 
    tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 
    ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer rdrand lahf_lm 3dnowprefetch epb pti 
    ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat md_clear
```

```bash
    [noscere@meg ~]$ sudo howdy version
    [sudo] password for noscere: 
    Howdy 2.6.1
```

It would appear, given the working snapshot, that my camera configuration is not the issue, and there is some kernel or CPU instruction that is failing when trying to add a face but not whilst generating the snapshot?

However, here is my camera info, just in case:
```bash
    # The path of the device to capture frames from
    # Should be set automatically by an installer if your distro has one
    device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:3:1.0-video-index0
```

```bash
    [noscere@meg ~]$ ls /dev/v4l/by-path/
    pci-0000:00:14.0-usb-0:3:1.0-video-index0  pci-0000:00:14.0-usb-0:3:1.0-video-index1
```
```bash
    [noscere@meg ~]$ v4l2-ctl --list-devices
    Lenovo EasyCamera: Lenovo EasyC (usb-0000:00:14.0-3):
            /dev/video0
            /dev/video1
            /dev/media0
```
```bash
    [noscere@meg ~]$ v4l2-ctl -d /dev/video0 --all
    Driver Info:
            Driver name      : uvcvideo
            Card type        : Lenovo EasyCamera: Lenovo EasyC
            Bus info         : usb-0000:00:14.0-3
            Driver version   : 5.15.78
            Capabilities     : 0x84a00001
                    Video Capture
                    Metadata Capture
                    Streaming
                    Extended Pix Format
                    Device Capabilities
...etc...
```

Nos78, 2022-12-03:
```bash
    [noscere@meg ~]$ sudo howdy clear
    No models created yet, can't clear them if they don't exist
    [noscere@meg ~]$ sudo howdy list
    Face models have not been initialized yet, please run:

            sudo howdy -U noscere add

    [noscere@meg ~]$ sudo howdy test
    Illegal instruction
    [noscere@meg ~]$ sudo howdy disable 0
    The disable option has already been set to false
```

Nos78, 2022-12-03:
Here is my PKGBUILD from the designated AUR build directory - current @ /var/tmp/pamac-build-noscere/howdy
```shell script
# Maintainer: Frank Tackitt <frank@tackitt.net>
# Maintainer: boltgolt <boltgolt@gmail.com>
# Co-Maintainer: Raymo111 <hi@raymond.li>
# Contributor: Kelley McChesney <kelley@kelleymcchesney.us>

pkgname=howdy
pkgver=2.6.1
pkgrel=2
pkgdesc="Windows Hello for Linux"
arch=('x86_64')
url="https://github.com/boltgolt/howdy"
license=('MIT')
depends=(
        'opencv'
        'hdf5'
        'pam-python'
        'python'
        'python-pillow'
        'python-dlib'
        'python-face_recognition'
        'python-face_recognition_models'
        'python-click'
        'python-numpy'
        'python-opencv'
)
makedepends=(
        'cmake'
        'pkgfile'
)
backup=('usr/lib/security/howdy/config.ini')
source=(
        "https://github.com/boltgolt/howdy/archive/v${pkgver}.tar.gz"
        "https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2"
        "https://github.com/davisking/dlib-models/raw/master/mmod_human_face_detector.dat.bz2"
        "https://github.com/davisking/dlib-models/raw/master/shape_predictor_5_face_landmarks.dat.bz2"
)
sha256sums=('f3f48599f78fd82b049539fcfc34de25c9435cad732697bdda94e85352964794'
            'abb1f61041e434465855ce81c2bd546e830d28bcbed8d27ffbe5bb408b11553a'
            'db9e9e40f092c118d5eb3e643935b216838170793559515541c56a2b50d9fc84'
            '6e787bbebf5c9efdb793f6cd1f023230c4413306605f24f299f12869f95aa472')

package() {
        # Installing the proper license files and the rest of howdy
        cd "${pkgname}-${pkgver}"
        install -Dm644 LICENSE "${pkgdir}/usr/share/licenses/${pkgname}/LICENSE"
        mkdir -p "${pkgdir}/usr/lib/security/howdy"
        cp -r src/* "${pkgdir}/usr/lib/security/howdy"
        cp "${srcdir}/dlib_face_recognition_resnet_model_v1.dat" "${pkgdir}/usr/lib/security/howdy/dlib-data/"
        cp "${srcdir}/mmod_human_face_detector.dat" "${pkgdir}/usr/lib/security/howdy/dlib-data/"
        cp "${srcdir}/shape_predictor_5_face_landmarks.dat" "${pkgdir}/usr/lib/security/howdy/dlib-data/"
        chmod 600 -R "${pkgdir}/usr/lib/security/howdy"
        mkdir -p "${pkgdir}/usr/bin"
        ln -s /lib/security/howdy/cli.py "${pkgdir}/usr/bin/howdy"
        chmod +x "${pkgdir}/usr/lib/security/howdy/cli.py"
        mkdir -p "${pkgdir}/usr/share/bash-completion/completions"
        cp autocomplete/howdy "${pkgdir}/usr/share/bash-completion/completions/howdy"
}
```

dmig, 2022-12-12:
@Nos78 this issue is not related to `howdy` in any way. From this PKDBUILD you can see, package depends on `python-dlib`, which I believe installs incorrect dlib binary to your system.

To fix this, follow steps from my comment on the issue where you started posting earlier: [#101](https://github.com/boltgolt/howdy/issues/101#issuecomment-445518764)
And not just copy commands, but try to read and understand what's that comment about.

Likely, there is a way to set build options, so installed `python-dlib` will be custom-built for your CPU, but I'm not aware of that. Feel free to find it and create a pull request.

closed 2026-11-16 by boltgolt