boltgolt/howdy #732: Tests pass, but doesn

lucaliceti, 2022-12-04:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Howdy doesn't work for anything authentication related, such as running a command with sudo, or signing in to gnome 42.
Running both `sudo howdy test` and `sudo howdy snapshot` works.

But howdy test returns:
`[ WARN:0@0.070] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (2401) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.070] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (1356) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.070] global /build/opencv/src/opencv-4.6.0/modules/videoio/src/cap_gstreamer.cpp (862) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created`

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): 
-OS: Arch Linux x86_64
- Host: 81YM IdeaPad 5 14ARE05
- Kernel: 5.15.81-1-lts

Howdy version (`sudo howdy version`):
- Howdy 2.6.1

captjay98, 2023-02-13:
Hello ,did you manage to get a fix?

yi-ge, 2023-03-09:
I have the same problem.

boltgolt, 2023-03-09:
I see you're on arch, have you added any lines to the PAM?

lucaliceti, 2023-03-10:
Could you explained what you mean by PAM? I've never hear of that before. Thank you.

boltgolt, 2023-03-10:
It's what regulates how authentication is done on the system. Did you use the arch wiki guide to install Howdy? That includes setting the right settings in PAM files

musikid, 2023-10-13:
Closing this as it seems to be a misconfiguration problem. Feel free to reopen it if you don't agree.

closed 2023-10-13 by musikid