boltgolt/howdy #740: Unable to install howdy for Ubuntu 22.04 due to *.bz2 files.

aahlad2000, 2022-12-31:
Facing Issues for installation of Howdy for Ubuntu 22.04. Apparently error message says it can't read .bz2 files.
These are the following commands I have used :
**sudo add-apt-repository ppa:boltgolt/howdy
sudo apt update
sudo apt install howdy**

[image]

Batwam, 2023-01-02:
Could this be a duplicate of this issue? https://github.com/boltgolt/howdy/issues/738

Batwam, 2023-01-02:
for info, I have run the [dlib-data/install.sh](https://github.com/boltgolt/howdy/blob/beta/howdy/src/dlib-data/install.sh) without problem:
[image]

Does it run for you?

aahlad2000, 2023-01-07:
Unable to run this command.
Screenshot from 2023-01-07 16-49-38.png

aahlad2000, 2023-01-07:
Unable to run this command
[image]

aahlad2000, 2023-01-07:
Unable to run this command

file:///home/aahlad_kethineedi/Pictures/Screenshot%20from%202023-01-07%2016-49-38.png

aahlad2000, 2023-01-07:
Unable to run this command.
[image]

Batwam, 2023-01-07:
Yeah, I didn't meant type it as a command, it was just a link to the file. I meant download the script and run it.

aahlad2000, 2023-01-08:
Still unable to run it.
[image]

Batwam, 2023-01-09:
This is pretty odd as the script is pretty simple and uses wget to download 3 files. It even checks if wget is installed and falls back on `curl` of `wget` isn't installed. I though perhaps you are somehow missing `wget`  or `curl` or even `hash`. You should be seing some sort or error...

Are you able to do the manual download of the 3 files? 
`wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2`
`wget https://github.com/davisking/dlib-models/raw/master/mmod_human_face_detector.dat.bz2`
`wget https://github.com/davisking/dlib-models/raw/master/shape_predictor_5_face_landmarks.dat.bz2`

If so you could just download them, leave them there (don't extract) and run the main code again. This way it should see them and extract when it gets to the extracting phase. I'm sure this is going to solve the main issue as there could be other issues in the code but it's worth a try...

aahlad2000, 2023-01-21:
I see.
However I am unable to download the 3 files. I got this error :
failed: Network is unreachable.

aahlad2000, 2023-01-21:
Also, looks like I am unable to access these URLs from the browser as well.

Batwam, 2023-01-22:
You could give it a try. I had a look and you would need to save the files in `/usr/lib/security/howdy/dlib-data`. 

I don't know how you manage to get a different ppa but using the same commands. The ppa for this package you send you to `http://ppa.launchpad.net/boltgolt/howdy/ubuntu/dists/j` and `jammy is in the Release list.

Note that I also just tried the ppa installation method on a fresh 22.10 install in a VM and it's spitting out quite a few warnings...
[image]

3gsahil, 2023-06-03:
> This is pretty odd as the script is pretty simple and uses wget to download 3 files. It even checks if wget is installed and falls back on `curl` of `wget` isn't installed. I though perhaps you are [...]

unable to download the files altogether .

Batwam, 2023-06-03:
Feels like some sort of proxy issue as the files are available at the locations provided as far as I can tell. Are you using some sort of school/work/public network which may be blocking file storage perhaps?

EricMuChR, 2025-01-31:
Same problem for me. Also Ubuntu 22.04. Just couldn't find a way to install those 3 files. Is there any hope that this problem will be resolved if I update the system version to like Ubuntu 24.04?

closed 2026-11-16 by boltgolt