boltgolt/howdy #742: Nothing has happend on DebiaN

zsxboli, 2023-01-07:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

The installation process went smoothly.And sudo howdy test is good running.Everything is OK.But It does not seem to work.
In terminal,type "sudo  -i",I also need to enterpasswd.More than that,in login interface and other scenarios that require password input,it does not work!

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):Debian Testing+Gnome

Howdy version (`sudo howdy version`):2.6.1

boltgolt, 2023-01-09:
Did you run `sudo howdy add`?

zsxboli, 2023-01-12:
> 

Ye.I am ready for everything.

zsxboli, 2023-01-12:
> Did you run `sudo howdy add`?

> > 
> 
> Ye.I am ready for everything.

I had reinstalled my system,and then ,reinstalled howdy with gdebi.And did as https://github.com/boltgolt/howdy/issues/711#issuecomment-1306559496  (but on debian ,the 'pam.py' in /lib/security/howdy,so,I modified the path).

And then ,I did as https://github.com/boltgolt/howdy/issues/734#issuecomment-1349218211 ,modified the pam.py.And  sudo chmod +x /lib/security/howdy/pam.py.

Finally ,I successed.All running well.

vmlinux26, 2023-05-05:
mine is running fine except for the screen unlock after the computer has been in sleep mode from closing laptop
running 'mate' 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"

zsxboli, 2023-11-18:
> mine is running fine except for the screen unlock after the computer has been in sleep mode from closing laptop
> running 'mate' 
> PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
> 
> edit "chmod +x /lib/security/howdy/pam.py"  doesn't work on mine

Maybe this problem is related to linux desktop environment.
I tried this measure on Gnome and succeed.

Jung1ePeng, 2024-04-15:
> Maybe this problem is related to linux desktop environment. I tried this measure on Gnome and succeed.

After following the instructions here, everything worked fine. I also use Debian testing with GNOME. Before this, I referred to the [Arch Wiki](https://wiki.archlinux.org/title/Howdy) and add this to the sudo and gdm-password files:

>auth sufficient pam_unix.so try_first_pass likeauth nullok

And that makes a mistake for me, so just remove it, then everything works fine. It seems to be a difference between Arch and Debian.

closed 2024-04-20 by musikid