boltgolt/howdy #747: dlib error

galen1423, 2023-02-04:
Hi,

Since this morning, after updates daily system **Ubuntu 23.04** (Lunar Lobster), **howdy** brings **dlib** error.

`Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 17, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
` 

I tried to install **dlib** - (by the way why isn't it installed anymore..?), but that doesn't solve the situation...  after install **cmake**, **dlib** fails to compile. Not sure that is **howdy** issue but if you have an idea..?

Linux distribution (Ubuntu 23.04):

Howdy version (2.6.1)

Thanks for your work

All-Ki, 2023-03-15:
I fixed this by cloning the dlib repo , doing pip install cmake and then pip setup.py install in the dlib folder
Hope that helps

All-Ki, 2023-03-16:
Same for me , here is the full output log ( kali 2023.1 python 3.11 )
[log.txt](https://github.com/boltgolt/howdy/files/10990079/log.txt)

paolocalosso, 2023-04-13:
same here, dlib cannot compile under ubuntu 23.04 with python3.11

closed 2026-11-16 by boltgolt