boltgolt/howdy #751: "All frames were too dark"

EternalHighway, 2023-02-20:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

I've been trying to add my face using howdy on Linux Mint, but every time I do, the terminal says that the frames are too dark. I've tried in different lighting situations, but they all have the same result.
It usually says that the average default is 70-80, when the threshold is 50.
Any support would be appreciated.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Mint (latest version)

Howdy version (`sudo howdy version`): latest version

boltgolt, 2023-02-20:
Have you ran `sudo howdy test` to see if your frames really are dark? If not, you can up how dark a frame has to be to be rejected in the config

EternalHighway, 2023-02-21:
>https://github.com/boltgolt/howdy/issues/751#issuecomment-1436481884

After running the test, the resulting camera footage is extremely dark, colorless, and flashing. My webcam works fine in other cases, so I'm assuming I didn't properly configure howdy with it. How could I have it work as intended?

boltgolt, 2023-02-22:
Are you trying to set up a IR or normal camera?

EternalHighway, 2023-02-22:
> Are you trying to set up a IR or normal camera?

I'm on a Dell XPS15 9500, so it's an IR camera.

PabloRuizCuevas, 2023-03-05:
I'm having also an XPS and in may case I had to select manually one of the 4 options I had the others were too dark.

go to: ` sudo howdy config ` and check out which device are you using
then go to:
`/dev/v4l/by-path`

to see which devices are available and just try all of them changing the path in the config and using `sudo howdy test`

EternalHighway, 2023-03-07:
> 

After checking the 4 options, 1 was too dark, 2 led to sudo howdy test failing, but one was bright enough, but colorless.
I'm not sure if that's supposed to happen, but thank for the advice!

closed 2023-03-09 by boltgolt

EternalHighway, 2023-03-09:
> Colorless is the indication that you've found the right camera! IR cameras have no color of course

It still won't recognize my face though. And the red sensors aren't turning on at all.

boltgolt, 2023-03-10:
The red dots are actually IR light emitters, you might be able to get them to work with [the great tool](https://github.com/EmixamPP/linux-enable-ir-emitter) from @EmixamPP 

After camera configuration, did you clear and readd your models? Is your PAM setup for Howdy to fire?