boltgolt/howdy #752: Is it possible to disable howdy requirement for login?

sneakyjoeru, 2023-02-22:
It looks like now my laptop requires me to enter password and have my face in sight. Can't login just with password. If my face gets damaged, I imagine I won't be able to login?
I modified 2 files and used chmod as described in this post: https://github.com/boltgolt/howdy/issues/219#issuecomment-674606116
How can I make Howdy optional?

boltgolt, 2023-02-22:
Howdy is optional by default. It should activate first and if you're not detected it should fall back to the password prompt. Are you sure that does not happen?

sneakyjoeru, 2023-02-22:
When I boot system I see my password prompt. Whenever I hit Enter - be it with or without password, Howdy activates and scans. 
It won't let me in unless a face is matching (for instance if I enter a correct password and cover webcam with my hand) - will shake password field and keep it filled.

On a lockscreen it behaves differently.
With the password it will let me in just fine.

Without the password it can switch to a new screen without password but with an Unlock button. 
Then it starts scanning again and if it doesn't see the face it gets stuck on that screen until I start mashing space, enter and escape. If it scans the face, it switches to a same screen again, but this time I can press enter to login or wait until it scans the face again and log me in automatically.

I'm on kubuntu 22.10, if that matters.

boltgolt, 2023-02-22:
Probably best for you to not use it on login with these screensaver bugs. You could enable Howdy only for things like sudo [with this how-to](https://github.com/boltgolt/howdy/wiki/Only-using-howdy-for-specific-authentication-types)

closed 2023-02-22 by boltgolt