boltgolt/howdy #755: No recognition on Popos 22.04, howdy 2.6.1

jontis, 2023-02-24:
No recognition of face.
Added two face models and then test in exact same conditions.
sudo howdy test shows diagnostics and a clear face with a red circle encircling face including chin but without hair.'
Seems reasonable. Moving around a bit to give the camera different tilts, angles and faces, but mostly trying to look exactly like when model was added.
The circle never turns green.
Tried fast and slow mode.
Tried certainty settings from 1 to 9.

This is for Popos 22.04 (debian)
Howdy 2.6.1
The hardware is a laptop system76 Lemur lemp9. To my understanding, this is not an IR capable camera, but lighting was good and I was under the impression that it should work then.

boltgolt, 2023-02-24:
More of an UX issue i think, the test command in 2.6.1 only shows what faces have been recognized, not if they match a model. Referring to #504 again for whenever 3.0.0 sees the light

closed 2023-02-24 by boltgolt

jontis, 2023-02-25:
Well, yes and no. While your comments clarify the UI, I used the test function because I've never had howdy match a face.
I'll hang on until that code is released. Honestly, I thought it already was as I read the code to understand what the function was doing. Didn't occur to me to check if the source code I was reading was in the version I was using ;-) Anyhow, if you could give a short direction how to run the source, that would be appreciated. I'm decent in python, but didn't figure out how to test the app with code changes locally. Kindly

boltgolt, 2023-02-27:
The easiest might be to fully uninstall Howdy and install the [latest build of the beta package (as of 27-01-2023)](https://launchpad.net/~boltgolt/+archive/ubuntu/howdy-beta/+files/howdy_3.0.0-beta+35_amd64.deb) but i'm not entirely sure how stable that is

jontis, 2023-02-27:
Many thanks. I'll try that. Still curious if you could recommend a way to run the source code as I might be able to contribute some code myself :-)

boltgolt, 2023-02-27:
When developing i install howdy normally, then delete `/lib/security/howdy` and clone a fresh git copy there, that should usually do it. As always with developing a PAM module, keep a terminal as root open so you can't be locked out

jontis, 2023-02-28:
Thank you! I'll be careful. 
Out of curiosity, if I mess up, will ssh into the machine still work so I can repair it that way?