boltgolt/howdy #759: Nothing happens

gaziroffka, 2023-02-28:
So. I've installed Howdy AUR from software center. Everything was smooth.
Configured everything. All was smooth aswell.
Tested. Added face. Smooth.
Added those recommended lines to /pam.d/
And I do not see a single blink of an emitter when I am asked for password. What can be wrong?
Just nothing happens.
Garuda linux / xfce de

peterroth, 2023-03-09:
Did you configure Howdy with the `sudo howdy config` command?

gaziroffka, 2023-03-09:
Yep. And all was good> Did you configure Howdy with the `sudo howdy config` command?

peterroth, 2023-03-09:
Did you check if something is recorded in journalctl at the time when you try to use Howdy? Additionally, you can set Howdy (with the config file) to debug mode, so it will print out into a log file what's the problem.

gaziroffka, 2023-03-10:
> 

> Did you check if something is recorded in journalctl at the time when you try to use Howdy? Additionally, you can set Howdy (with the config file) to debug mode, so it will print out into a log file what's the problem.

What do you mean by term 'using'? Configure is done and has no issues, and this is all i can do with Howdy, because nothing else happens!

peterroth, 2023-03-10:
for example is you run `sudo howdy test`, does the window open and shows the picture of the cam? What does journalctl records when you run that command?

gaziroffka, 2023-03-10:
> for example is you run `sudo howdy test`, does the window open and shows the picture of the cam? What does journalctl records when you run that command?

window does open and camera works just fine, I have added faces.
command:

sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
^C
Closing window

I'm not so good with linux, and am not familiar with journalctl - what exactly i must command?

gaziroffka, 2023-03-10:
> Great troubleshooting by @peterroth so far!
> 
> Where are you trying to activate Howdy? Have you tried using it with sudo?

I want Howdy to scan my face when password prompted. But Howdy does not activate, even though i added faces etc. Howdy works only to and add/test. That's all where it is active

boltgolt, 2023-03-10:
Right, and if you run `sudo -i` you only get asked for a password and no other output

gaziroffka, 2023-03-10:
> Right, and if you run `sudo -i` you only get asked for a password and no other output

exactly! 
sudoing, logging in, updating app, etc etc - Howdy does not start

muhamed1ism, 2023-04-14:
I have same issue

ipa1981, 2023-04-20:
In a new Ubuntu 23.04 I also facing the same. Everything set up, but no launching on sudo or login. Probably something related to PAM

NickFlamel, 2023-04-22:
This worked for me: https://github.com/boltgolt/howdy/wiki/Howdy-on-Debian-Testing---KDE-Plasma#fix-the-script-for-debian-testing

ipa1981, 2023-04-23:
> This fixed the issue for me: https://github.com/boltgolt/howdy/wiki/Howdy-on-Debian-Testing---KDE-Plasma#fix-the-script-for-debian-testing

This worked for me. Thanks @NickFlamel

kiraitachi, 2023-05-06:
@NickFlamel  thanks mate that worked for me too on Kali Linux!

AuraPy, 2024-04-26:
@NickFlamel Thank you SO much! Worked on Debian 12 LTS!

closed 2026-11-16 by boltgolt